SevenBits/Mac-Linux-USB-Loader #59: Do you Vette what you code?

juanitajones, 2015-01-03:
Please forgive my question, but do you vette what your coding? I ask this because it's painfully obvious that the "majority" of the the people that are creating a USB with different Linux iterations are not booting in  neither Mavericks or Yosemite. It's not that I'm not great full for your effort, but when 85% of members cant boot into Kali, (or any other Linux) ......that's a problem.

jotaemei, 2015-01-03:
"I ask this because it's painfully obvious that the "majority" of the the people that are creating a USB with different Linux iterations are not booting in neither Mavericks or Yosemite."

SevenBits is providing this to the community for free.  Your criticism based on unfounded assumptions is not cool.  Why even put the word majority in quotes?  Is it because you have no idea actually how many people this is not working for?  

"but when 85% of members cant boot into Kali, (or any other Linux) ......that's a problem."

From where do you get this number 85?  And claiming that 85% of members can't boot into "any other Linux" is beyond the pale.

I've noticed that it's rare for people who are having difficulty booting to specify which line of Macs they're using. I can boot into Kali from a LiveUSB created by Mac Linux USB Loader just fine on a Mac with new architecture.  I can't, however, use the LiveUSB on older Macs.  A known problem is attempting to boot from USBs on older Macs, and SevenBits explained that here: https://github.com/SevenBits/Mac-Linux-USB-Loader/issues/30#issuecomment-24002841 (among a few places)

kalihack2014, 2015-01-03:
@juanitajones @jotaemei  I agree with jotaemei!! SevenBits does good work for basically nothing and is knowladeable with computers. 

@jotaemei How were you able to boot into Kali Linux via Live USB???? I would really like to know how you got it to work!

juanitajones, 2015-01-03:
I'm using 3 different MacBook pros from 4 years ago to 9 months ago. Three running Yosemite and one running Mavericks. I got one mavericks machine to boot to Ubuntu 10.4 (not 14.10) That's it.
The USBs are being created in both VM's and actuals of Yosemite and Mavericks. You tell me..........?

kalihack2014, 2015-01-03:
@jotaemei  I thought you said you got Kali Linux to work. I can get Ubuntu 12.04.2 and 14.04.1 to work no problem those are the only two I've tried but they work. I just can't get Kali Linux to work via Live USB.

juanitajones, 2015-01-03:
I've tried kali x32 and x64, no luck.

kalihack2014, 2015-01-03:
@juanitajones  I bet it's the boot in Blind Mode problem you are having? It's notorious in Kali...

juanitajones, 2015-01-03:
I'm on another machine right now, but if I remember correctly it was another error message.

kalihack2014, 2015-01-03:
@juanitajones Then probably the failed to boot debian message if you used enterprise...

juanitajones, 2015-01-03:
I'll check later or tomorrow. Has none been successful booting either Kali or Tails?

kalihack2014, 2015-01-03:
@juanitajones I know one person who was able to boot Kali Linux via USB on a PC but none that I've heard of on a Mac. I don't know really anything about Tails.

juanitajones, 2015-01-03:
I have everything booting on my windows machines no problem. Never a problem of any kind using Easy2Boot.

kalihack2014, 2015-01-03:
@juanitajones Yep... I hear Kali will work on a PC no problem.

SevenBits, 2015-01-03:
@juanitajones Issues like what you all are having happen in every software product. Do you know how many people file support requests with Apple on the day a new OS X release comes out? I've been told it's a quite substantial number. I understand your frustration, but it's impossible with my limited resources to test every single use case - I have created dozens of live USBs using my tool on Yosemite and all work. Have you looked in the logs to find any error messages Mac Linux USB Loader might have printed? You also, as others have pointed out, neglected to give any information on your system. (Does nobody read the official rules regarding posting bug reports?)

kalihack2014, 2015-01-03:
@SevenBits  so you got Kali Linux to work?

SevenBits, 2015-01-03:
@kalihack2014 Some internal tests that I'm doing look promising, but nothing rock solid yet... I will try to release a beta ASAP. Debian is less friendly to UEFI than Ubuntu, hence the trouble everyone's had with Kali (and Tails).

kalihack2014, 2015-01-03:
@SevenBits Very Nice, Very Nice! I can't wait for the release and hope you can get Kali Linux to work on a bootable USB :)  That would be fantastic.

jotaemei, 2015-01-03:
"@jotaemei How were you able to boot into Kali Linux via Live USB???? I would really like to know how you got it to work! I have a Mac 11,3 Mid 2014 and can't get Kali Linux to boot." - @kalihack2014 

Ah.  Ancient Mac Linux USB Loader secret.  A wise user once said,
[image]
 "Are you using the Mac Linux USB loader 3.0? If you are don't it doesn't work at all lol" https://github.com/SevenBits/Mac-Linux-USB-Loader/issues/57#issuecomment-68226176

Explains all. lmaooo

jotaemei, 2015-01-03:
OK.  All the machines here are running Mavericks, updated to the latest version (10.9.5).  My understanding, though perhaps only @SevenBits can confirm, is that it doesn't matter what version of OS X you have installed when attempting to boot from the LiveUSB.  The version of OS X you have, I'd assume, is only relevant when you're creating the LiveUSB - if Mac Linux USB Loader (MLUL) calls into different versions of OS X system libraries.

Anyway, I made the LiveUSB on a Mac Book Pro from mid-2009.  The result of attempting to boot into Kali on that MBP is falling into blind mode and then seeing a screen of horizontal striped lines.  It's completely unusable, and I assume it's because of the lack of support for the nvidia drivers.  Booting into Kali from a LiveDVD, as @SevenBits explained elsewhere, works on this old 2009 machine, except for some glitches here and there like the bottom half of the pointer being missing and Kali freezing up the machine after logging out of Gnome  - though this can be resolved by upgrading your Kali distribution if you install it to your hard drive.

How I got a Kali LiveUSB to work
--------------------------------------------
* USB flash drive of 4GB
* DiskUtility partitioned USB to "MS-DOS (FAT)", Options: "Master Boot Record"
* Kali Linux 1.0.9a 64-bit ISO (kali-linux-1.0.9a-amd64.iso) downloaded from https://www.kali.org/downloads/ (with the sha sum and certificate confirmed as explained on that page)
* and probably most importantly, Mac Linux USB Loader 2.0.2
* booted on an iMac13,1

There is something important to note though.  Booting to Kali (and Ubuntu) on this iMac screwed up the pairing of the Mac wireless keyboard and the magic trackpad.  So, I wasn't able to log in.  It was wicked obnoxious too.  In order to repair the input devices with the Mac, it was necessary to reboot the Mac a few times and reconfigure the keyboard.  So, I'm done with testing on that iMac.

I'll going to run an experiment on a recent MBP and tell you what the results were in a few minutes.  I expect better results since there are no input devices to worry about pairing.

I'm thinking that perhaps since there's a way to get a working LiveUSB of Kali for recent Macs, @SevenBits, should leave up version 2 of MLUL for people to download and then mark version 3 as "experimental" or something.

jotaemei, 2015-01-03:
Results of testing on a MacBookPro11,1:

[image]

With this test, I encountered the problem mentioned elsewhere that the credentials

user:root
password:toor

were not accepted.

I've got some things to do now and don't have time to look up that other thread about credential problems, but at least there's some promising news about being able to boot into Kali.  I'll look into the other issues later.

kalihack2014, 2015-01-03:
@jotaemei Yes I know if you use an older version of Mac Linux USB Loader you can boot into blind mode, very known problem.

SevenBits, 2015-01-03:
@juanitajones @jotaemei For all of those having issues with missing configuration files... when using version 3.0.1, try manually creating the EFI and BOOT folders on your USB stick before running Mac Linux USB Loader. This comes from a suggestion from an email I received from someone who read this thread; apparently, the installation worked after that.

I'm sure that the Kali solution will be working soon... until then, try to contact me from my site directly with support requests; this thread is getting long and it is not intended to be a help forum.

jotaemei, 2015-01-03:
"@jotaemei Yes I know if you use an older version of Mac Linux USB Loader you can boot into blind mode, very known problem."

@kalihack2014, so why aren't you doing it that way?  Mac Linux USB Loader 3.0 is a hot mess.

gorilla1977, 2015-01-04:
@jotaemei @juanitajones After solving this issue by using SevenBits's advice with folders, I got into
another issue - which is slightly different, and also related to configuration file. For those who interested, please lets move there, because the situation has changed and this thread is indeed getting too long:
http://github.com/SevenBits/Mac-Linux-USB-Loader/issues/60

SevenBits, 2015-01-04:
I'm going to close this because this is a chat thread, and not a true issue.

closed 2015-01-04 by SevenBits

SevenBits, 2015-03-08:
@jotaemei @kalihack2014 Hey guys, I'm wondering if you'd like to try out my new beta for Kali. I believe that maybe the login will work now, but I don't have access to hardware which Kali can boot on so I'm stuck actually trying a graphical login.

Since the version of Enterprise with 2.0.2 came closest to being able to boot Kali, my solution uses the version of GRUB which shipped with that release, but uses the newset Enterprise.

The download link is [here](http://sevenbits.tk/downloads/), and if you really need it, here's the [direct link](http://sevenbits.tk/downloads/kali-test.tar.gz).