SevenBits/Mac-Linux-USB-Loader #60: [3.0.1] Configuration problems

gorilla1977, 2015-01-04:
I have tried SevenBits's advice about creating folders ( *mentioned at* [*thread 1*](http://github.com/SevenBits/Mac-Linux-USB-Loader/issues/58) and [*thread 2*](http://github.com/SevenBits/Mac-Linux-USB-Loader/issues/58) )
It seems I moved one step further, because this time the configuration file was created.
But, when I tried to boot, I got a new error message, which is slightly different :

*Error: configuration file parsing error.*
Cannot continue because core files are missing or damaged.

My .MLUL-Live-USB contains the following:

    #This file is machine generated. Do not modify it unless you know what you are doing.
    
    entry Other
    family Other

Of course I haven't edited it, it was automatically generated by MLUL.
Entry and family are "Other" because I had to select "Other" at "Distribution Family"
*(Qubes OS is based on Fedora, so it is wrong to choose Ubuntu or Debian)*

kalihack2014, 2015-01-04:
@gorilla1977  so you got a MLUL file... What system were you using Mavericks or Yosemite?

gorilla1977, 2015-01-04:
@kalihack2014 I'm using Yosemite, but I'm sure there would have been the same results with Mavericks

kalihack2014, 2015-01-04:
@gorilla1977  I'm kind of at the same stage you are because that is what my MLUL file says and I created it with Yosemite except I used version 3.0. I wouldn't be so sure about the same result in Mavericks because I used 3.0.1 in Mavericks and it did not create a MLUL file, regardless I doubt it's the MLUL file causing the trouble but it could be. I know in Kali there is some UFEI problems according to what sevenbits told me....

kalihack2014, 2015-01-04:
Try updating your MLUL file to this, I'm about to try it...
entry Kali
family Debian
kernel /live/vmlinuz boot=live noconfig=sudo username=root hostname=kali

kalihack2014, 2015-01-04:
Nope doesn't work with Kali Still get this screen...
[image]

gorilla1977, 2015-01-04:
@kalihack2014 I've made tests earlier, and the problem occurred both on Mavericks and Yosemite.
After trying this advice, the problem disappeared at Yosemite, so I expect it to disappear on Mavericks
as well. I haven't observed any difference in test results of 3.0.1 between Yosemite and Mavericks, yet

kalihack2014, 2015-01-04:
@gorilla1977  trying it out now, we will see if it creates a MLUL file. 3.0.1 Mavericks. I'm reformatting the drive now.

gorilla1977, 2015-01-04:
@kalihack2014 *In past I have experienced the same problem as you - "i8042: No controller found"
That's the issue of Debian, not of MLUL. [Check my Ubuntu bug report](https://bugs.launchpad.net/ubuntu/+bug/1406832) Try adding "nomodeset", it should solve your problem

kalihack2014, 2015-01-04:
@gorilla1977 are you trying to create a bootable USB for Ubuntu? And what do you mean try adding "nomodeset"?

gorilla1977, 2015-01-04:
@kalihack2014 3 days ago I needed to boot Ubuntu 14.10 from USB. It supports UEFI so it boots on Macbook Pro without any additional "EFI loaders", but only if I add "nomodeset" to kernel booting options

kalihack2014, 2015-01-04:
@gorilla1977  I don't know what you mean lol
But as far as creating a MLUL with 3.0.1 on Mavericks there is none created (simple issue to fix by creating your own in VIM,NANO, or Textedit) But here is a screenshot just to show you and I have hidden files show.

[image]

kalihack2014, 2015-01-04:
@gorilla1977 what do you mean nomodeset to the booting options, how do you do that?

gorilla1977, 2015-01-04:
@kalihack2014  I am going to give you an Ubuntu example, you need to somehow apply it to Debian.
During the booting process, I had to edit `grub.cfg` entry for "Try Ubuntu without installing".
By default, it contained:
	set gfxpayload=keep
	linux	/casper/vmlinuz.efi  file=/cdrom/preseed/ubuntu.seed boot=casper quiet splash --
	initrd	/casper/initrd.lz
I changed it into:
	set gfxpayload=keep
	linux	/casper/vmlinuz.efi  file=/cdrom/preseed/ubuntu.seed boot=casper quiet splash nomodeset --
	initrd	/casper/initrd.lz
After that it boots successfully. But of course your case is different, and syntax should be different too.
Try booting with `kernel /live/vmlinuz boot=live noconfig=sudo username=root hostname=kali nomodeset`
if it does not help, you should try a different syntax of appending "nomodeset" to kernel booting options

kalihack2014, 2015-01-04:
@gorilla1977  okay that makes more sense however how were you able to get into/edit the grub.cfg file?

gorilla1977, 2015-01-04:
@kalihack2014 Booting process got me into GRUB2 menu, where I was able to edit "Try Ubuntu without installing" entry. It threw me into GRUB2 menu automatically... (you may have to press something)

kalihack2014, 2015-01-04:
@gorilla1977  okay I know about Grub... so you just had to like press tab to edit some files or something?
Kali doesn't use grub so I don't know exactly how to edit that and it doesn't even boot into anything that I can edit.

gorilla1977, 2015-01-04:
@kalihack2014 After I select Ubuntu's "EFI Loader" at Mac's UEFI menu, it shows me "Booting in insecure mode" message, then it throws me into GRUB2 menu, where I could do three things:
* go Up/Down with arrows, highlight menu entry, and then press "Enter" (thats what most users do)
* press "e" key to edit the GRUB2 entry, then save it and use for booting
* press "c" key to go to GRUB2's command line (e.g. for booting of ISO file from hard drive)

By the way, I've just learned that its possible to add `nomodeset` option by pressing 2 in Enterprise and then just choosing it. That is more user-friendly way, and it should work for you

Thank you for hinting about extra "kernel" line, that helped me to get past the "parsing" error to Enterprise menu. But my case is not closed yet, because I need to correctly guess that "kernel" line...

Do you know if there should be another "initrd" line in MLUL file, just under the "kernel" line

kalihack2014, 2015-01-04:
@gorilla1977  okay I know exactly what you mean now, I just tried it in a VM but in order for grub to boot you have to enable EFI.

Now what do you mean by "Thank you for hinting about extra "kernel" line, that helped me to get past the "parsing" error to Enterprise menu. But my case is not closed yet, because I need to correctly guess that "kernel" line..." I don't think I hinted you about anything. sorry for so many questions I am learning here

gorilla1977, 2015-01-04:
@kalihack2014 its okay ;) Just to clarify the things: in my case with Ubuntu I didn't have to "enable EFI", use MLUL, etc. - it was semi-bootable right out-of-the-box after using "dd" command to write ISO to USB

My MLUL had two important lines: entry and family.
But your comment "Try updating your MLUL file to this" - features 3 lines! - and 3rd line is "kernel line" :
`kernel /live/vmlinuz boot=live noconfig=sudo username=root hostname=kali`
I have no idea where you got this 3rd line. It is obvious that my 3rd line should be different.
And I wonder if there could/should be a 4th line which features `initrd`

kalihack2014, 2015-01-04:
@gorilla1977  I got that script from another user on github he told me to try it. Hope you can use it to try to get Kali Linux to boot via Live USB :)

And about enable EFI it's because I was using it in Virtual Box, I know in real life booting the USB would automatically take you to GRUB2.

gorilla1977, 2015-01-04:
@kalihack2014 @SevenBits Small updates:
1) I could get to Enterprise menu only if I choose family as Debian.
If I would try to set it as Other, like it was after MLUL file generation, I would get an error
"Distribution Family "Other" is not supported"
2) If I would choose Debian, it will get to Menu, but later it will try to get the kernels at /live directory of ISO . But Qubes (or Fedora 17/18) have a different ISO structure.
They don't have `/live` and their `kernel` & `initrd` are at different places: `/isolinux/` or `/images/pxeboot/`

kalihack2014, 2015-01-04:
Well here is my screenshot after using Mac Linux USB Loader 3.0.1 on Mavericks Kali Linux Distro...

[image]

gorilla1977, 2015-01-04:
@kalihack2014 Have you created `efi` and `efi/boot` directories, before running MLUL?

SevenBits, 2015-01-04:
@gorilla1977 You can specify the locations of the kernel and initrd in the configuration file, and you do not need to specify a family. Just do something like this:

    entry Qubes OS
    kernel ...
    initrd ...
    root ...

Fill in the dots with the proper locations.

kalihack2014, 2015-01-04:
@gorilla1977 Yes Mac Linux USB Loader did that for me I have a folder called efi and inside that folder is boot then inside that are the boot.efi, bootx64.efi and the boot.iso. There is however no MLUL file. Version 3.0.1 on Mavericks doesn't create one.

kalihack2014, 2015-01-04:
@gorilla1977  However I created a .MLUL-Live-USB with the text: 
"entry Kali
family Debian
kernel /live/vmlinuz boot=live noconfig=sudo username=root hostname=kali"
And it gets me to the Boot failed screen.

kalihack2014, 2015-01-04:
@SevenBits  Alright I understand most of your post trying to edit for Kali Linux:
entry Kali Linux
kernel /live/vmlinuz
initrd /live/initrd.img
root ... (I don't know what to put here) 

Also I did some research my initrd.img is only 22.1 MB However my boot.iso file is 3.05gb but the three folders inside that iso (efi, install, and live) only equal up too about 62mb. Makes me wonder where all the other data is. Thinking about using hdiutil convert source.iso -format UDRW command to convert my kali iso to a img and replacing the initrd.img with that... What do you think?

SevenBits, 2015-01-04:
@kalihack2014 The file `initrd.img` does not do what you think... it serves as a basic file system which can be read by the kernel at boot time and contains things like drivers for the root (`/`) partition. Boot splash screens are also contained in here.

The `root` directive in Enterprise's configuration file is, in simplistic terms, usually the folder containing the kernel and initrd. In your case, it would be `live`.

kalihack2014, 2015-01-04:
@SevenBits  Thanks, I will give it a try.

kalihack2014, 2015-01-04:
@SevenBits  Edited the MLUL file to this:
"entry Kali Linux
kernel /live/vmlinuz
initrd /live/initrd.img
root /live"

And it just gives me a failed boot screen...

kalihack2014, 2015-01-05:
@SevenBits  do you have an idea about why kali is not able to boot? Any progress? I think it uses GRUB instead of GRUB2 if that helps any...

Thanks for your time SevenBits! I know you are busy.

SevenBits, 2015-01-05:
@kalihack2014 It's probably an issue with pointing the system to its bootable ISO file - once the kernel and RAM disc are loaded, the kernel needs to hand off control to the regular system on the root partition (mounted as `/`). This works with Ubuntu, but it doesn't work with Debian (and, by extension, Kali). Interestingly, this works on the version of Enterprise included in version 2.0.2 of Mac linux USB Loader, but in the version included in 3.0, it does not, even though the code for booting the ISO is essentially the same.

The Kali entry in my sample Enterprise configuration file on [Enterprise's website](https://sevenbits.github.io/Enterprise/) contains the proper parameters for booting Kali. Thus, as soon as this issue is resolved, I see no reason why Kali and other Debian distributions such as Tails won't work.

SevenBits, 2015-01-05:
Also, @gorilla1977, I believe that in the latest revision I've fixed the issue regarding the configuration file not being created. The only question is would the community prefer a fix a) now, but without a fix for Kali or b) later, when Kali works

I'm assuming that you all will prefer the first option, but do let me know.

gorilla1977, 2015-01-05:
@SevenBits Please consider a "quiet release" : quietly replace the existing 3.0.1 version, with a 3.0.1 version that includes a fix.
1) Those who had a problem with missing config, already know a solution (create efi & boot folders)
2) Even if someone doesn't know, it would be easy to look it up (these issues are 1st result in Google)
3) New downloaders won't need this solution at all, and they could live without knowledge of that issue
So this approach looks harmless.
After checking the version history of MLUL, I see that previous releases brought a lot of improvements.
Small fix with config file, it just doesn't look sufficient for 3.0.2 release, if there is nothing to add

SevenBits, 2015-01-05:
@gorilla1977 The problem with that approach is that it breaks semantic versioning: there would be effectively *two* different versions Mac Linux USB Loader 3.0.1 in the wild, with no obvious way to tell them apart (yes, you could use the build number, but regular users won't think to check that). A disadvantage of doing this is that users who already downloaded the current version and whose USB doesn't work will not think to download the same version again. Even if they did, it seems a little much to ask users to search on Google for a solution to a problem.

kalihack2014, 2015-01-05:
@SevenBits  I personally would rather have Kali to work :)

gorilla1977, 2015-01-06:
@SevenBits well its to append letter to version number: of course not `a` or `b` (because that would mean "alpha" or "beta") but `f` - which means `fixed`. So it would be `3.0.1f` ;) its still much better than something like `3.0.1.5` - such long version numbers aren't looking nice

gorilla1977, 2015-01-06:
@kalihack2014 Alternative way of trying to launch Kali linux:
**1)** Make your MLUL file to look like the following:

    #This file is machine generated. Do not modify it unless you know what you are doing.

    entry Kali
    kernel /isolinux/isolinux.bin

Obviously, it will fail, but instead of failed Debian booting (which you are experiencing now) it will fail at earlier stage and you will be taken to GRUB2 menu (GRUB2 is embedded to Enterprise) There, you could run a following command:

`syslinux_configfile -i /isolinux/isolinux.cfg -r /`

It will ask syslinux to boot Debian in a isolinux-style. It could work, but I haven't tested it yet.

kalihack2014, 2015-01-06:
@gorilla1977  Okay I will try it, although of course I have one question what do you mean by 
syslinux_configfile -i /isolinux/isolinux.cfg -r /

Is that like editing the file you showed me in Ubuntu or actually running this command in a grub shell?

gorilla1977, 2015-01-06:
@kalihack2014 It is for running this command in a GRUB shell, which is bundled with Enterprise.
Of course, with the same success you could run Ubuntu LiveUSB from a second flash drive, go to GRUB2 command line by pressing `c`, running commands `loopback loop (MLUL USB here)/efi/boot/boot.iso`, `set  root=(loop)`, then run the `syslinux_configfile` command from above. But this way you will be forced to have two USB ports locked. Macs used to have only a few USB ports, and thanks to that MLUL includes GRUB2 shell you would be able to use just one USB port instead of two

kalihack2014, 2015-01-06:
@gorilla1977  don't exactly know what you mean with Ubuntu but for Kali if I get prompted for the GRUB2 shell I will run this command 
"syslinux_configfile -i /isolinux/isolinux.cfg -r /" exactly how it is written and let you know what happens... fingers crossed.

gorilla1977, 2015-01-06:
@kalihack2014 Open the iso file of Kali Linux (with archiver, or by mounting it), then go to `isolinux` folder and open `isolinux.cfg` file. You'll see that it uses a menu.cfg, which in turn uses various configs such as `live.cfg` and `install.cfg`, which are providing the exact commands to syslinux for booting Kali Linux

gorilla1977, 2015-01-06:
@kalihack2014 -r flag is optional, its for telling the root folder. I am not sure if it should be `/` or `/isolinux`

kalihack2014, 2015-01-06:
@gorilla1977  it said:

error: no video mode activated.
grub>

gorilla1977, 2015-01-06:
@kalihack2014 Hmmm... Well, you could use those `isolinux` config files from above as "cheatsheets",
and try to use the commands from them in MLUL config file. Or (as an alternative) you could make a copy of that isolinux.cfg file, modify it, then put it somewhere at MLUL USB and execute "syslinux_configfiles` on it in a similar fashion

kalihack2014, 2015-01-06:
@gorilla1977  I have no idea what you are talking about.... I'm trying to follow what you are saying but don't.

gorilla1977, 2015-01-06:
@kalihack2014 pls refresh a page, I have removed the difficult alternative xD

kalihack2014, 2015-01-06:
@gorilla1977  here is my email since this is starting to get into a "support forum"

kylecmvp@gmail.com

gorilla1977, 2015-01-06:
@kalihack2014 ok I will write you a quick message there (a disadvantage of this is that other users won't be able to read our experiences)

kalihack2014, 2015-01-06:
@gorilla1977  I understand, but SevenBits said so and I don't want to get suspended on here lol

gorilla1977, 2015-01-06:
lets remove some of our unnecessary messages here, to reduce a flood

gorilla1977, 2015-01-06:
lets remove some of our latest unnecessary messages here, to reduce a flood. and its not a good idea to post your email openly without style like kylecmvp [att] gmail [dottt] com (otherwise, spam bots will harass you)

kalihack2014, 2015-01-06:
Okay I removed them and my email

SevenBits, 2015-01-07:
FYI, the thread is being closed because the reason for the error is that Enterprise cannot boot USBs with a family of `Other`. The expectation is that if your chosen distribution is not supported then you will go in manually after the installation is done and set it up manually.

closed 2015-01-07 by SevenBits

gorilla1977, 2015-01-08:
@SevenBits @kalihack2014 @bsfa Small update about booting Qubes R2, for those who are interested: [Qubes-Devel : UEFI + Qubes R2 booting problems on Macbook Pro 8,2](http://groups.google.com/forum/#!topic/qubes-devel/NAUIW79Vfxc) In short, its **impossible** to boot Qubes R2, because Xen older version doesn't support EFI, but could be possible to boot new Qubes R3 version after it will come out. Qubes case is frozen & will not be mentioned until a new R3 version comes

SevenBits, 2015-03-08:
@jotaemei @kalihack2014 Hey guys, I'm wondering if you'd like to try out my new beta for Kali. I believe that maybe the login will work now, but I don't have access to hardware which Kali can boot on so I'm stuck actually trying a graphical login.

Since the version of Enterprise with 2.0.2 came closest to being able to boot Kali, my solution uses the version of GRUB which shipped with that release, but uses the newset Enterprise.

The download link is [here](http://sevenbits.tk/downloads/), and if you really need it, here's the [direct link](http://sevenbits.tk/downloads/kali-test.tar.gz).

robsalasco, 2015-11-26:
any news on this? I want to boot qubes in my macbook!