SevenBits/Mac-Linux-USB-Loader #61: USB drive not showing up

enzocagggiano, 2015-01-04:
When I try to create a live usb, my usb drive doesn't show up. I made sure its plugged in, tried reformatting it, erasing it with different filesystems, but nothing has worked. The usb drive shows up fine in finder and everywhere else (and in older versions of this app). I am using the latest version.

kalihack2014, 2015-01-04:
@enzocagggiano  This is a known issue. The USB drive has to 4gb or less and formatting it to smaller partitions does not work either. Sevenbits told me about this.

SevenBits, 2015-01-04:
This appears to be an OS X problem; I have tried looking into this in the past with no clear solution.

enzocagggiano, 2015-01-04:
Ok, will it be fixed soon?

enzocagggiano, 2015-01-04:
Ok, hope it is fixed soon. A 4GB usb drive works?

kalihack2014, 2015-01-04:
@enzocagggiano  It worked for me but if you are trying to boot a Kali Linux distribution or Qubes I don't think anyone has been successful, except to boot Kali in Blind Mode which is completely useless...

enzocagggiano, 2015-01-05:
I'm trying to boot ubuntu

kalihack2014, 2015-01-05:
@enzocagggiano 
http://www.ubuntu.com/download/desktop/create-a-usb-stick-on-mac-osx

http://computers.tutsplus.com/tutorials/how-to-create-a-bootable-ubuntu-usb-drive-for-mac-in-os-x--cms-21253

These have worked for me perfectly!

gorilla1977, 2015-01-05:
@enzocagggiano Actually, if you want to boot Ubuntu 14.10 you don't need MLUL (or any other EFI loaders)
Just "dd" the ISO image to your USB drive, and it should work right out of the box! ;-)

enzocagggiano, 2015-01-05:
I get a no controller found error

gorilla1977, 2015-01-05:
@enzocagggiano You need to add `nomodeset` to kernel booting options.
While booting, click `e` key to edit the entry for "Try Ubuntu without installing".
`By default, it contains:`
*set gfxpayload=keep*
*linux /casper/vmlinuz.efi  file=/cdrom/preseed/ubuntu.seed boot=casper quiet splash --*
*initrd /casper/initrd.lz*
`You need to change it into:`
*set gfxpayload=keep*
*linux /casper/vmlinuz.efi  file=/cdrom/preseed/ubuntu.seed boot=casper quiet splash* **nomodeset** *--*
*initrd /casper/initrd.lz*
After that it boots successfully.
**I already wrote about solution above in this thread - https://github.com/SevenBits/Mac-Linux-USB-Loader/issues/60 **

enzocagggiano, 2015-01-05:
I got to the boot screen and the busybox window but now I get this error:
(Initramfs) Unable to find a medium containing a live file system

gorilla1977, 2015-01-05:
@enzocagggiano That's very strange... Tell me please:
**1)** What is your hardware version of Mac? (that could be important)
**2)** What steps did you follow to create a bootable USB? What commands have you used?

enzocagggiano, 2015-01-05:
1. MacBook Pro 17 Inch Late 2011
2. I used the enterprise files. I get to the normal ubuntu boot menu after vmlinuz kernel issues. I can't get the USB drive to boot with dd

gorilla1977, 2015-01-05:
@enzocagggiano Could you show me the exact "dd" command that you have tried?
My hardware is similar to yours ( MBP 15" Early 2011 ) , but "dd" method worked for me at 1st try!

enzocagggiano, 2015-01-05:
What did you format the USB drive to

enzocagggiano, 2015-01-05:
sudo dd if=~/ubu.img of=/dev/rdisk1 bs=1m

kalihack2014, 2015-01-05:
@enzocagggiano  did you convert the iso to a dmg? it looks like you are using an img

enzocagggiano, 2015-01-05:
it has to be a dmg? it says to make a img on the official guide.

kalihack2014, 2015-01-05:
@enzocagggiano  you have to convert the iso to a dmg via 
"hdiutil convert -format UDRW -o ~/path/to/target.img ~/path/to/ubuntu.iso"

then 
sudo dd if=/Users/johnnywinter/Downloads/ubuntu-14.04-desktop-amd64+mac.dmg of=/dev/rdisk2 bs=1m

Notice that the the ubuntu file is a .dmg 

In the official guide the not says "Note: OS X tends to put the .dmg ending on the output file automatically." 

and step 8 says:
Execute

sudo dd if=/path/to/downloaded.img of=/dev/rdiskN bs=1m
(replace /path/to/downloaded.img with the path where the image file is located; for example, ./ubuntu.img or ./ubuntu.dmg). <-------- here .dmg

enzocagggiano, 2015-01-05:
ok

kalihack2014, 2015-01-05:
@enzocagggiano  yes OSX uses dmg files not img lol should work now! :)

enzocagggiano, 2015-01-05:
ok, I will tell you if i get an issue

kalihack2014, 2015-01-05:
@enzocagggiano then do what gorilla1977 said when you get to GRUB2

"You need to add nomodeset to kernel booting options.
While booting, click e key to edit the entry for "Try Ubuntu without installing".
By default, it contains:
set gfxpayload=keep
linux /casper/vmlinuz.efi file=/cdrom/preseed/ubuntu.seed boot=casper quiet splash --
initrd /casper/initrd.lz
You need to change it into:
set gfxpayload=keep
linux /casper/vmlinuz.efi file=/cdrom/preseed/ubuntu.seed boot=casper quiet splash nomodeset --
initrd /casper/initrd.lz
/\/\/\
After these changes it boots successfully ;) If you want them by default, edit grub.cfg on USB drive.
I already wrote about the nomodeset solution above in this thread"

enzocagggiano, 2015-01-05:
GOT IT TO WORK!

closed 2015-01-05 by enzocagggiano

kalihack2014, 2015-01-05:
@enzocagggiano  Knew you would ;)

SevenBits, 2015-01-05:
@kalihack2014 @enzocagggiano FYI, guys, in the future, please avoid discussing software other than Mac Linux USB Loader. If you want to give each other support, please use an Internet forum or email each other privately.

enzocagggiano, 2015-01-05:
ok

kalihack2014, 2015-01-05:
@SevenBits Will do

ParamaShiva1, 2023-01-07:
It is 2023 now and I am having this issue now. Can somebody please look into this. I have tried even a 2GB drive.