SevenBits/Mac-Linux-USB-Loader #62: Error message: "…the Enterprise source that you have selected is either incomplete or missing."

nhoizey, 2015-01-07:
I downloaded Mac-Linux-USB-Loader today (v3.0.1), used it to download Unbuntu 14.10 and Mint 17.1, but both gave me this message after clicking the "Begin Installation" button:

[image]

I'm using OS X 10.9.5 on a 2014 13" Macbook Air.

SevenBits, 2015-01-07:
Hello, what Enterprise source are you using? What does the Enterprise version selector say?

nhoizey, 2015-01-07:
I'm using "Included With Application" v0.2.1:

[image]

SevenBits, 2015-01-07:
@nhoizey Please show me the logs for Mac Linux USB Loader... also give me a layout of the files in the .app.

nhoizey, 2015-01-07:
The logs:

```
07/01/2015 13:01:20,008 lsboxd[315]: Not allowing process 85552 to launch "/Applications/Mac Linux USB Loader.app" because it has not been launched previously by the user, 
07/01/2015 13:01:20,008 lsboxd[315]: Not allowing process 85552 to register app "/Applications/Mac Linux USB Loader.app" for launch.
07/01/2015 13:01:20,240 Mac Linux USB Loader[85552]: Couldn't find dictionary of Enterprise source file locations. Is this the first run? Creating one now...
07/01/2015 13:01:20,267 Mac Linux USB Loader[85552]: Detected eligible volume at /Volumes/Nifty. Type: hfs
07/01/2015 13:01:20,268 Mac Linux USB Loader[85552]: Detected eligible volume at /Volumes/SANS TITRE. Type: msdos
07/01/2015 13:01:20,269 Mac Linux USB Loader[85552]: Detected eligible volume at /Volumes/LINUX. Type: msdos
07/01/2015 13:01:55,083 Mac Linux USB Loader[85552]: Invalid color System, labelColor (warning given only once)
07/01/2015 13:02:09,879 Mac Linux USB Loader[85552]: Updating current downloads table.
07/01/2015 14:09:08,711 Mac Linux USB Loader[85552]: view service marshal for <NSRemoteView: 0x7fdccb5570c0> failed to forget accessibility connection due to Error Domain=NSCocoaErrorDomain Code=4099 "Couldn’t communicate with a helper application." (The connection was invalidated from this process.) UserInfo=0x7fdccb60d880 {NSDebugDescription=The connection was invalidated from this process.}
timestamp: 14:09:08.711 Wednesday 07 January 2015
process/thread/queue: Mac Linux USB Loader (85552) / 0x109c29000 / com.apple.NSXPCConnection.user.endpoint
code: line 2972 of /SourceCache/ViewBridge/ViewBridge-46.2/NSRemoteView.m in __57-[NSRemoteView viewServiceMarshalProxy:withErrorHandler:]_block_invoke
domain: communications-failure
 
07/01/2015 14:09:31,922 Mac Linux USB Loader[85552]: Sparkle: ===== Mac Linux USB Loader.app =====
07/01/2015 14:09:53,812 Mac Linux USB Loader[85552]: Updating current downloads table.
07/01/2015 14:37:13,485 Mac Linux USB Loader[85552]: Warning: NSActivity <_NSActivityAssertion: 0x7fdccb4d1a00> was ended multiple times
07/01/2015 15:18:48,694 launchservicesd[137]: Application App:"Mac Linux USB Loader" asn:0x0-1e6be6a pid:85552 refs=7 @ 0x7fbcd4a2d3a0 tried to be brought forward, but isn't in fPermittedFrontApps ( ( "LSApplication:0x0-0x1001 pid=126 "loginwindow"")), so denying. : LASSession.cp #1481 SetFrontApplication() q=LSSession 100006/0x186a6 queue
07/01/2015 15:18:48,694 WindowServer[157]: [cps/setfront] Failed setting the front application to Mac Linux USB Loader, psn 0x0-0x1e6be6a, securitySessionID=0x186a6, err=-13066
07/01/2015 15:35:10,166 Mac Linux USB Loader[85552]: view service marshal for <NSRemoteView: 0x7fdccd8b9c00> failed to forget accessibility connection due to Error Domain=NSCocoaErrorDomain Code=4099 "Couldn’t communicate with a helper application." (The connection was invalidated from this process.) UserInfo=0x7fdccd901f40 {NSDebugDescription=The connection was invalidated from this process.}
timestamp: 15:35:10.166 Wednesday 07 January 2015
process/thread/queue: Mac Linux USB Loader (85552) / 0x109c29000 / com.apple.NSXPCConnection.user.endpoint
code: line 2972 of /SourceCache/ViewBridge/ViewBridge-46.2/NSRemoteView.m in __57-[NSRemoteView viewServiceMarshalProxy:withErrorHandler:]_block_invoke
domain: communications-failure
 
07/01/2015 15:37:40,402 Mac Linux USB Loader[85552]: view service marshal for <NSRemoteView: 0x7fdccb7e8d20> failed to forget accessibility connection due to Error Domain=NSCocoaErrorDomain Code=4099 "Couldn’t communicate with a helper application." (The connection was invalidated from this process.) UserInfo=0x7fdccb66a000 {NSDebugDescription=The connection was invalidated from this process.}
timestamp: 15:37:40.402 Wednesday 07 January 2015
process/thread/queue: Mac Linux USB Loader (85552) / 0x109c29000 / com.apple.NSXPCConnection.user.endpoint
code: line 2972 of /SourceCache/ViewBridge/ViewBridge-46.2/NSRemoteView.m in __57-[NSRemoteView viewServiceMarshalProxy:withErrorHandler:]_block_invoke
domain: communications-failure
 
07/01/2015 15:38:33,322 Mac Linux USB Loader[85552]: view service marshal for <NSRemoteView: 0x7fdccd8a4dc0> failed to forget accessibility connection due to Error Domain=NSCocoaErrorDomain Code=4099 "Couldn’t communicate with a helper application." (The connection was invalidated from this process.) UserInfo=0x7fdccd8750c0 {NSDebugDescription=The connection was invalidated from this process.}
timestamp: 15:38:33.322 Wednesday 07 January 2015
process/thread/queue: Mac Linux USB Loader (85552) / 0x109dbd000 / com.apple.NSXPCConnection.user.endpoint
code: line 2972 of /SourceCache/ViewBridge/ViewBridge-46.2/NSRemoteView.m in __57-[NSRemoteView viewServiceMarshalProxy:withErrorHandler:]_block_invoke
domain: communications-failure
 
07/01/2015 15:53:38,590 Mac Linux USB Loader[85552]: Updating current downloads table.
07/01/2015 16:39:47,867 Mac Linux USB Loader[85552]: Warning: NSActivity <_NSActivityAssertion: 0x7fdccb4d1a00> was ended multiple times
07/01/2015 16:46:21,440 Mac Linux USB Loader[85552]: view service marshal for <NSRemoteView: 0x7fdccd880180> failed to forget accessibility connection due to Error Domain=NSCocoaErrorDomain Code=4099 "Couldn’t communicate with a helper application." (The connection was invalidated from this process.) UserInfo=0x7fdccd87e420 {NSDebugDescription=The connection was invalidated from this process.}
timestamp: 16:46:21.440 Wednesday 07 January 2015
[... cut ...]
./Resources/Base.lproj:
Acknowledgements.rtf                      SBEnterprisePreferencesViewController.nib
Credits.rtf                               SBGeneralPreferencesViewController.nib
MainMenu.nib                              SBPersistenceManagerWindowController.nib
SBAboutWindowController.nib               SBUSBSetupWindowController.nib
SBDocument.nib

./Resources/Enterprise:
boot.efi    bootX64.efi

./Resources/Tools:
mke2fs

./Resources/en.lproj:
InfoPlist.strings

./_CodeSignature:
CodeResources
```

raulnabarret, 2015-01-14:
@nhoizey Were you able to fix the issue?

nhoizey, 2015-01-14:
@raulnabarret no, I switched to a Windows computer with a live Ubuntu USB key instead… :-/

SevenBits, 2015-01-14:
@raulnabarret @nhoizey To fix this, quit Mac Linux USB Loader and try going to `~/Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/Data/Library/Application Support/` and deleting the file `EnterpriseInstallLocations.plist` (or whatever the name is) and restarting the app. A user told me in an email that this works. In the meantime, please don't move Mac Linux USB Loader around on your hard drive: I suspect that there is a problem updating the path of the bundled Enterprise version that ships with the app.

j0emay, 2015-01-18:
Couple issues here; may want to 'suggest' move to /Applications (fixes that error)

ref: https://github.com/potionfactory/LetsMove/

forked to look into this as well as FileVault-specific (I believe) errors.

SevenBits, 2015-01-18:
@j0emay Thanks, and I've heard of that library - doesn't work with sandboxed applications though, but I may be able to work around that.

andrenatal, 2015-01-22:
The delete of EnterpriseInstallationLocations.plist did not worked for me.

SevenBits, 2015-01-22:
Did you quit the application before you deleted the file?

andrenatal, 2015-01-22:
Yes.

SevenBits, 2015-01-23:
@andrenatal And are you using the latest version?

FYI, this quote from [a somewhat notable Cocoa developer](http://www.potionfactory.com/blog/2009/09/17/move-applications-folder) should be of interest to anyone wondering why this is happening::

> Zip files are great for distributing Mac software. It's an established format understood by all kinds of software, both Mac and otherwise. Some browsers will even decompress them automatically leaving one single application icon in your Downloads folder. Having seen a few people get confused by disk images and being inspired by Coda, I thought I'd give it a shot.

> The switch to Zip went great except for one thing: we started getting a wide variety of mysterious crash reports. They all had something to do with the app not being able to load essential resources such as nib files. A nib file contains data to create user interface elements such as windows, buttons, and so forth. Users aren't going to get very far with a Cocoa app if it can't load its nib files.

> It took me some time to connect the dots, but once I did, it was pretty damn obvious. People were moving The Hit List into the Applications folder while it is still running. Mac users think nothing of moving a file while it's being used. Mac OS X does some fancy things to allow this for the majority of cases so I wasn't expecting Cocoa to get tripped up by something that now seems trivial. But alas, it was trying to load files from the old location and freaking out when it couldn't find them.

So thus appears to be a Mac OS X problem. The newest update should fix this in most cases, but I'm still investigating ways of stomping this out fir good which are sandbox compatible.

Robto, 2015-01-25:
I have the same problem with 3.0.2 in combination with Ubuntu 14.0.4.1 & Ubuntu14-10 (downloaded via distributed download). Btw, where does it download the images to? It's not in my Downloads folder and the finder icon (or delete icon) are not working. I run the tool from the dl folder

Robto, 2015-01-25:
And moving it to applications results in a authorization screen in which clicking 'grant access' does not work.

[image]

SevenBits, 2015-01-25:
@Robto To find your downloaded ISO file, select the distribution, and then right click and choose the "Reveal in Finder" option.

andrenatal, 2015-01-26:
Yes, I was using the latest version.

SevenBits, 2015-01-26:
@andrenatal This is under investigation.

pf0, 2015-01-27:
have the same problem. wanted to make a elementary os live usb than this error came...
hope you will fix this soon :)

SevenBits, 2015-01-27:
@pf0 Did you ever move around Mac Linux USB Loader? Or did this error occur the first time you launched it from the Downloads folder?

wfrenzel, 2015-01-28:
Had the same problem!
I found a solution that worked for me: The problem was, that I've tried to run the Mac Linux USB Loader from the download folder. Doing so, resulted in the same error message described here. I "solved" it very eays: Just drag the program to the application folder (install the software) and than start it again - works like a charm :-)

C0t300, 2015-01-30:
It worked for me too! I drag it from the download folder to the application, restart it, and its working pretty fine!

SevenBits, 2015-01-30:
@wfrenzel @C0t300 Great, thanks for your discovery! If a few others can verify that this fixes the problem, I can add this to the public wiki so that everyone can see it.

pf0, 2015-01-30:
tried @wfrenzel 's solution and now it works :) thank you!

wfrenzel, 2015-01-30:
glad to read that my "solution" works for you guys to :-)

callmeskywalker, 2015-01-31:
@wfrenzel Moving it to the applications folder worked for me as well, thanks! (OS X 10.8.5)

SevenBits, 2015-01-31:
Alright, I will add this info to the wiki. Thanks guys.

commit 1f8a5090 (2015-05-30): Fixes #62

This commit fixes a bug whereby the Enterprise configuration database would not be valid until the first launch of the application, making the creation of a USB drive fail until the user restarts the app.
commit 39591b2e (2015-05-30): Fixes #62

This commit fixes a bug whereby the Enterprise configuration database
would not be valid until after the first launch of the application,
making the creation of a USB drive fail until the user restarts the app.
closed 2015-05-30 by SevenBits