SevenBits/Mac-Linux-USB-Loader #63: No .MLUL-Live-USB file when creating Elementary OS-Luna

kassi, 2015-01-07:
I downloaded the Elementary OS-Luna via Distribution Downloader and started to Create Live USB.
I didn't change the options the program chose, which is "Ubuntu" and a marked checkbox "This ISO has code older than Ubuntu 14.10".
After creating the USB I found out that there's no .MLUL-Live-USB file being created which was the case when I used a (manually downloaded) ubuntu 14.10+mac iso file.

I'm using the newest version 3.0.1 (8010).
The Enterprise version (I found after some time inside preferences) is: Included with Application 0.2.1.

What do I have to write to the file?

SevenBits, 2015-01-07:
This is a known issue and it has been discussed already in a separate thread. To cut to the chase, you either need to run the installation wizard again or create the `EFI` and `BOOT` folders manually before the wizard.

FYI, this was fixed already; I am just waiting for some translations to come through before I issue the update containing the fix.

kassi, 2015-01-07:
Ok, didn't realize this when I was searching for similar issues, because it was mentioned more as a side note.
I do now have a .MLUL file, but also other issues now:
* the content says `kernel /casper/vmlinuz.efi file=/cdrom/preseed/ubuntu.seed` but this file does not exist inside the mentioned iso (checked it). I had to remove `.efi` to be able to load the kernel.
* after loading I get a lot of errors:
    `udevd[135]: timeout: killing '/sbin/blkid -o udev -p /dev/sda1' [246]`

After some time I get the logo which disappears after some more time and brings me back to the command line.

    udevd[135]: '/sbin/blkid -o udev -p /dev/sda1' [246] terminated by signal 9 (Killed)
    [  144.277343] hda_intel: azx_get_response timeout, switching to single_cmd mode: last cmd=0x000f0001
    Welcome to elementary OS Luna (GNU/Linux 3.2.0-51-generic x86_64)
    
     * Website: http://elementaryos.org
    elementary@elementary:~$

SevenBits, 2015-01-13:
@kassi Elementary OS is based on Ubuntu 12.10. Therefore, you need to check the "contains legacy Ubuntu" checkbox in the ISO setup screen in Mac Linux USB Loader.

Some older Ubuntu distributions have many issues booting on Macs. I suggest either upgrading to Ubuntu 14.10 or wait for elementary OS to start using an update package base.

scerazy, 2015-02-26:
Any idea how .MLUL-Live-USB is supposed to look for Elementary iso to actually load?

sebus

SevenBits, 2015-02-26:
@scerazy For Ubuntu-based distributions, such as elementary, the internal layout of the ISO file is largely irrelevant because you can specify in the configuration file where the special files (such as the kernel and RAM disc) are located. Of course, the defaults as written by Mac Linux USB Loader are simply that - defaults - and you can customize them. You must also ensure that you give Mac Linux USB Loader the proper information about the distribution which you wish to load. This is all discussed in the tutorial.

And Enterprise itself can detect ISO files which are anywhere on your USB as long as they are in the `EFI/BOOT/` folder or a subfolder (due to a bug, at least one file must be named `boot.iso` - this is a regression from before Enterprise supported booting multiple distributions). You can specify the location using the `iso` parameter as detailed on the website.

Any suggestions to improve the documentation would be welcome.

scerazy, 2015-02-26:
Well on the site there are only parameters, not actual examples.

Is it

iso =
iso:
iso - 

Could not manage to boot Elementary with

kernel /casper/vmlinuz.efi file=/cdrom/preseed/ubuntu.seed 

Just died on not being able to find filesystem

Thanks

sc

closed 2015-05-29 by SevenBits