SevenBits/Mac-Linux-USB-Loader #64: [Kali Linux] mount: mounting /dev/sda2 on /live/medium failed: Invalid argument

gorilla1977, 2015-01-08:
I decided to file this as a separate issue, because I haven't seen anyone experiencing it yet.
During the process of booting Kali Linux LiveUSB, just after three pairs of those messages:

    sd 6:0:0:0: [sdb] No Caching mode page found
    sd 6:0:0:0: [sdb] Assuming drive cache: write through

I see this message which is being printed repeatedly for about a minute:

    mount: mounting /dev/sda2 on /live/medium failed: Invalid argument

Then I get the same error report like everyone else - `BOOT FAILED! This Debian Live image failed to boot."`

I've tried various kernel parameters, including those suggested at Enterprise website:
`kernel /live/vmlinuz boot=live noconfig=sudo username=root hostname=kali`

Also, I tried pulling the latest version of SevenBits from github repository and building it from source code (currently, it has a few commits ahead of released 3.0.1 version)

However that didn't help to solve a "mount problem". Do you have any ideas?

SevenBits, 2015-01-08:
Try the Debian mailing list. In the absence of a finished solution, they may be able to give you, the end users, some advice on this issue. Since Kali likely inherits its live USB setup procedure from Debian, this is worth considering.

Either that, or find an official email for someone at Offensive Security. Oddly, I can't seem to find one.

gorilla1977, 2015-01-08:
@SevenBits I wrote to `debian-live` mailing list, and would contact Kali through forum or bug report.
But, sadly, I am not sure if anyone would help with this very specific issue... Debian guys could tell "that's not our problem, contact Kali" and Kali will tell that USB booting at MBPs that are older than "late 2012" (that includes all non-retina MBPs) is not supported by them, as they state in their manuals & forum threads. Unless someone knowledgeable arrives to this thread, guess I'm out of luck :disappointed:

gorilla1977, 2015-01-10:
Small update: Debian Live developers refused to help and told me to contact Kali

SevenBits, 2015-04-03:
@kalihack2014 @Addisonbean @Soap109 If anyone is waiting for a progress report on Kali, you'll be happy to see these photos.

[image]
[image]

You'll have to wait for Mac Linux USB Loader 3.1, though...

reaganrockz, 2015-04-22:
When do you think Mac Linux USB Loader 3.1 will be released?

SevenBits, 2015-04-22:
@reaganrockz Sometime in the next week or two is the target.

reaganrockz, 2015-04-22:
@SevenBits Ok thanks!

SevenBits, 2015-05-01:
@reaganrockz A beta of version 3.1 is out... You can grab it on the releases page.

reaganrockz, 2015-05-01:
Ok thank you @SevenBits!

SevenBits, 2015-10-24:
Closing as version 3.1 has been released.

closed 2015-10-24 by SevenBits