SevenBits/Mac-Linux-USB-Loader #66: Linux Mint 17 : unable to find a medium containing a live file system

KeatsPeeks, 2015-03-05:
Hello.

I've been trying to install Linux Mint 17.1 x64 on my MBP mid-2014 running Yosemite, to no avail.
I followed the guide and tried several times with different options (all 4 combinations of the Ubuntu chekboxes), but it kept failing.

Just after the two initial menus, these errors appear : 
```
Loading Linux kernel...error: file /casper/vmlinuz.efi not found
Loading initial RAM disc...error: you need to load the kernel first
````

Then, a GRUB menu appears. I select the option to launch the OS and I can see the Linux Mint logo for a few seconds. 

Then, I get this fatal error : 
```initramfs : unable to find a medium containing a live file system```

Any ideas ? I don't know if the first two errors are related to the last one. Maybe I can enable some kind a debug log or try a few kernel switches ?
Thanks.

SevenBits, 2015-03-06:
@KeatsPeeks Hmm, the Linux Mint team must have changed something recently. I'll get on figuring out what's wrong.

For Linux Mint, you need to check **both** checkboxes on the setup screen. Did you do this?

KeatsPeeks, 2015-03-07:
@SevenBits Sorry, I guess I lied when I said I tried all combinations, I must have forgot this one. It works well when both checkboxes checked.
Sorry to have opened an issue on this for nothing. Keep up the good work.

closed 2015-03-07 by KeatsPeeks

cmivxx, 2015-06-15:
I am having the exact same problem on OSX Yosemite trying to load Mint 17.1.  The error persists whether I check both boxes or not on both the 3.0 version or the 3.1 BETA.  Any other ideas?  Any chance this would be alleviated by using Mint 17.0 or earlier?

SevenBits, 2015-06-16:
@cmivxx Can you include a screenshot of the error message?

cmivxx, 2015-06-16:
yeah, here you go.

http://imgur.com/a/1m9zy

SevenBits, 2015-06-16:
@cmivxx And the error occurs with Enterprise 0.3 (the one included with the most recent beta)?

Please try with Linux Mint 17... perhaps that will fix the problem...

cmivxx, 2015-06-16:
Will do..  I'll let you know how it goes.  Thanks for the help!

WAFFO, 2015-06-19:
So cmivxx didn't come back, but I can verify that using Linux Mint 17 instead of 17.1 works. I believe I checked both boxes as well.