SevenBits/Mac-Linux-USB-Loader #68: There is no such family 'Other'

YorikSar, 2015-04-03:
If one selects "Other" in "Distrubution Family" dropdown, Loader writes `family Other` line to `.MLUL-Live-USB` file and Enterprise complains that there is no such family.

SevenBits, 2015-04-03:
This behavior is mentioned on the [wiki page here](https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/General-Instructions). Quoth it:

> If you're installing an unsupported distribution, select Other for the distribution family and manually set up Enterprise yourself using the instructions here.

So, this isn't really a bug, but I'll admit that it could be annoying. I'm trying to find the best way to allow users to install unsupported distributions. I could ask for the location of the kernel, initrd, etc. from the GUI, but that might just confuse less skilled users. Do you have any suggestions?

YorikSar, 2015-04-03:
Yeah, makes sense. I think Loader should either ask for all options in user-friendly way or alert user that one should edit config file by hand (and probably open it). It would also be great if it already contained some skeleton with lots of comments describing what to write.

Another option would be to inspect ISO looking for `isolinux.cfg` (I guess most distributions use it) and parse it, it should be pretty straightforward if we hope that the first configuration that is found is the right one. It might be good to open generated config afterwards anyway for user to verify that everything went well.

SevenBits, 2015-10-24:
@YorikSar I am closing as issue as version 3.1 automatically opens the configuration file for the user if they choose an unsupported distribution, so this issue is somewhat mitigated now. I will, however, keep this feature in mind for future updates.

closed 2015-10-24 by SevenBits