SevenBits/Mac-Linux-USB-Loader #69: No localiza el USB (Fedora para Macbook Pro)

Bitluser, 2015-04-17:
Yo soy nuevo en todo esto. Quiero arrancar un USB con Fedora y para eso he utilizado Mac Linux USB Loader. 
El problema es que no me reconoce mi USB, ya probe con varios y cuando hice los primeros pasos como indicarle donde esta la imagen viva, continuo con la parte donde tengo que seleccionar el USB. Pero no me aparece nada, le doy en Refrescar pero sigue sin ubicar mi USB.

Aqui les dejo una captura de pantalla.
https://cloud.mail.ru/public/2p9sCLuUBxcW/errorubb.png

Saludos desde México

SevenBits, 2015-04-17:
Hola, ¿has intentado resolver el problema por formatear el USB usando Utilidad de Discos? Es probable que el USB no está formateado correctamente, y típicamente esto es una causa de varios problemas debido a el hecho que los Macs no reconocen un USB a menos que es formateado como FAT32 (u HFS+, pero esto no es recomiendo).

Bitluser, 2015-04-18:
Hola!
Muchas gracias por responderme, despues de publicar mi problema por este medio segui investigando e intentando buscar una solución. Al final termine haciendo lo siguiente.
Volvi a formatear mi USB con la opción Mac OS plus (con registro) despues reinicie mi Mac. Volvi abrir Mac Linux USB pero seguía el problema y no aparecia mi USB.
Total, que termine por formatear mi USB pero ahora con la opción MS-DOS (FAT), ya que recordaba algo sobre esta opción que era para que lo reconociera windows. (algo así)
Volvi abrir Mac Linux USB y oh!! sorpresa ahora si lo reconocio. Continue con los pasos y ahora si pudo crear mi USB con FEDORA :)
Pero..
El USB ya esta, pero el problema ahora viene cuando quiero instala Fedora.
Reinicio mi Macbook Pro y cuando hace el típico sonido de encendido dejo presionado el boton Alt para que me de opciones de discos. Si aparece mi USB con Fedora, lo selecciono y entonces la pantalla se pone en negro y se empieza a ejecutar. Aparecen varios números y solo alcance a leer algo que decia error y pum! se cancela y se inicia con normalidad mi Mac :(

Respuesta a SevenBits:
Hola amigo, gracias por tu respuesta. Bien, te comento que probe con otras memorias que ya estaban formateadas y aún así no me los reconoce. Como comentaba anteriormente solo cuando cambie la opcion de formatear a MS-DOS (Fat) es cuando ya lo reconocio. Pero ahora cuando intento instalarlo me manda un error :I  y se reinicia de forma normal.
Ahora ese es mi problema, yo se que este foro es para problemas con respecto a Mac Linux USB, pero en verdad deseo ya cambiarme a Fedora, e intentado buscar informacion sobre instalar Fedora en Mac en español y no he encontrado casi nada. Solo la página oficial de Fedora, pero pues no tiene traducción al español completamente :I 

Agradezco mucho que se tomen el tiempo para responderme.
Esto de GitHub me parece muy bueno.
Saludos y gracias.
Aqui algunas capturas de pantalla sobre lo que contiene el USB con Fedora con la opción MS-DOS

https://cloud.mail.ru/public/cujLuss4dZoR/1USB.png
https://cloud.mail.ru/public/3rRA1zRF8s4q/2USB.png
https://cloud.mail.ru/public/2hrQLPue1KiC/3USB.png

closed 2015-10-24 by SevenBits