SevenBits/Mac-Linux-USB-Loader #74: translation link is not found in homepage

Akagi201, 2015-05-18:
https://raw.githubusercontent.com/SevenBits/Mac-Linux-USB-Loader/stable/TRANSLATIONS
I digg the source code, this link is dead.

I'm glad to give a Chinese translation of the software and document. You can try <https://crowndin.com>. It's total free for open source projects. <https://crowdin.com/page/open-source-project-setup-request>

I am the Chinese translator of obs-studio software <https://crowdin.com/project/obs-studio>. The website is very friendly, easy to use.

SevenBits, 2015-05-18:
Thanks @Akagi201, and yeah, I've been planning on creating a new file soon; I'm aware that the link is dead. I'm designing a new webpage for Mac Linux USB Loader so that should be there around the same time.

I'd be happy to have you translate. I will look into that website.

SevenBits, 2015-10-24:
Closing as the website has been updated.

closed 2015-10-24 by SevenBits