SevenBits/Mac-Linux-USB-Loader #76: The Enterprise source that you have selected is either incomplete or missing

j-manu, 2015-05-29:
Sorry for the duplicating #62 but the issue happens on v3.1 preview #2. I have only run it from the Applications folder.  In the logs I see this

```
29/05/15 10:21:23.149 pm Mac Linux USB Loader[1400]: Couldn't find dictionary of Enterprise source file locations. Is this the first run? Creating one now...
```

After restarting the app

```
29/05/15 10:26:37.906 pm Mac Linux USB Loader[1400]: Detected eligible volume at /Volumes/Transcend. Type: msdos
29/05/15 10:26:37.926 pm Mac Linux USB Loader[1400]: view service marshal for <NSRemoteView: 0x7fc092f6e470> failed to forget accessibility connection due to Error Domain=NSCocoaErrorDomain Code=4099 "Couldn’t communicate with a helper application." (The connection was invalidated from this process.) UserInfo=0x7fc092ca0f30 {NSDebugDescription=The connection was invalidated from this process.}
timestamp: 22:26:37.926 Friday 29 May 2015
process/thread/queue: Mac Linux USB Loader (1400) / 0x10c7c0000 / com.apple.NSXPCConnection.user.endpoint
code: line 2972 of /SourceCache/ViewBridge/ViewBridge-46.2/NSRemoteView.m in __57-[NSRemoteView viewServiceMarshalProxy:withErrorHandler:]_block_invoke
domain: communications-failure
 ```

I'm on 10.9.5 iMac 21.5inch Mid 2011

SevenBits, 2015-05-29:
The second set of log messages don't appear to be related to the issue at all. You should see a message like this somewhere:

    		Found dictionary of Enterprise source file locations.

Can you restart the app once more? Usually it works after launching the app twice.

j-manu, 2015-05-30:
It worked after launching it once more :) Closing this

closed 2015-05-30 by j-manu

hovancik, 2015-08-04:
had the same issue just now. restarting app fixed that as well. 

mac 10.10.4, latest ver  mlul

jeremy-bruns, 2015-08-14:
same here 10.10.4 (14E46) restarting the app fixed it