SevenBits/Mac-Linux-USB-Loader #77: [FR] MacBook Air Mid 2014 - Tail Live USB

jcanonym, 2015-06-03:
ok so french keyboard.
strange bug, didn't work with a (more or less, not the exact sentence) strange error message as follow:
```invalid path: /live/wmlinu```
so in `USB Setup > Edit Entreprise Configuration File` 
i change this : `kernel /live/vmlinuz`
to this : `kernel /live/vmlinuzz`
and it worked !

SevenBits, 2015-06-03:
This bug looks like it should filed under SevenBits/Enterprise.

What version of Enterprise and Mac Linux USB Loader are you using?

jcanonym, 2015-06-03:
Mac Linux USB v.3.1 rev 9048.
For the Entreprise the one bundle with ML USB.

SevenBits, 2015-06-10:
Hmm, this is weird, but this issue has been reported before. I'll look into it.

SevenBits, 2015-10-24:
This issue has been fixed as of version 3.1, so I am closing this issue.

closed 2015-10-24 by SevenBits