SevenBits/Mac-Linux-USB-Loader #78: Problem with latest Kali version

agsola, 2015-06-19:
I have tried using Mac Linux USB Loader with the latest .iso from Kali (kali-linux-1.1.0a-amd64) but I couldn't make it work.

I attach two images with the error I get when booting.

Any idea?

First, it detects it:
[image]

But then, it throws an error:
[image]
It doesn't find the kernel, which I think is wrong (missing a "z"??).

SevenBits, 2015-06-19:
What model MacBook do you have?

agsola, 2015-06-22:
MacBook Air (13-inch, Mid 2013)

SevenBits, 2015-06-23:
Please [contact me](https://sevenbits.github.io/contact.html), as I want to send you an updated build with some changes so that I can see if they fix these issues for you.

powlwoq, 2015-06-29:
Hi 7bits,
i have the same issue with a MacBook Air (11-inch, 2015 [bought last month]).
Error msgs are the same like above with newest USB-Loader.
I can't even boot from external DVD - strange...
No chance to boot with newset rEFInd bootloader either (USB or DVD, 32 or 64bit).
I am really frustrated about this...
And I am thankful for any help out there.
Thanks in Advance,
powl

SevenBits, 2015-06-30:
@powlwoq Please contact me, as I want to send you an updated build with some changes so that I can see if they fix these issues for you.

SevenBits, 2015-07-06:
To anyone else having this issue, as a temporary workaround add an extra character to the end of the kernel file name in Enterprise configuration file (either .MLUL-Live-USB or enterprise.cfg, depending on which version of Mac Linux USB Loader you used).

SevenBits, 2015-09-14:
This was fixed in the latest release so I am closing this issue.

closed 2015-10-24 by SevenBits