SevenBits/Mac-Linux-USB-Loader #79: Cannot find the ISO /efi/boot/boot.iso

skylar73, 2015-06-24:
I have recently tried to use Mac Linux USB Loader to put Ubuntu 15.04 (64 bit) onto my flash drive. When booting, it will show the ubuntu screen (with it loading), but then it will go to this error message:
[image]

This same issue was brought up here: https://github.com/SevenBits/Mac-Linux-USB-Loader/issues/12 , but was never fixed. 

Additional details (that may have no relevance):
-Installed with persistence (max amount of GB, around 4.7 I think?)
-Booted and installed on a Mac with OS X Version 10.9.5
-Mac is a desktop from 2007

Thanks for your help!

SevenBits, 2015-06-24:
Which version of Mac Linux USB Loader are you using?

FYI, this should be filed under SevenBits/Enterprise.

skylar73, 2015-06-24:
3.0.2

SevenBits, 2015-06-24:
Can you try the 3.1 beta and see if that fixes it?

skylar73, 2015-06-24:
I did not try the 3.1 beta, but when I tried again on a different computer (Newer MacBook Pro, maybe mid 2013?), it worked. 

Thanks so much for your help and great program!!

closed 2015-06-24 by skylar73