SevenBits/Mac-Linux-USB-Loader #80: Macbook2,1 2007 won't recognize the USB drive

pizzamaker, 2015-07-01:
These are the specs: http://www.everymac.com/systems/apple/macbook/specs/macbook-core-2-duo-2.0-white-13-mid-2007-specs.html

When I boot holding down Option, I'm only offered to option to boot into OS X, no thumb drives show up.

I've done some digging and I think this might be the cause: https://wiki.archlinux.org/index.php/MacBook#Mid_2007_13.22_-_Version_2.2C1

But the instructions are very unclear. Can someone lend a hand?

SevenBits, 2015-07-04:
If your Mac has 32-bit EFI firmware, then it is not supported. This is stated very clearly on the homepage in the requirements section beneath the Download button.

closed 2015-07-04 by SevenBits

pizzamaker, 2015-07-04:
How can I go about?