SevenBits/Mac-Linux-USB-Loader #81: Download Location

ThePixelPony, 2015-08-10:
I've downloaded Linux Mint through the downloader but I do not know where the file is located and the magnifying glass icon does nothing.

Any help will be very much appreciated!

SevenBits, 2015-08-10:
You don't say which version you're using. You should be able to right click on the distribution's name and click Show in Finder or similar when the download has completed.

ThePixelPony, 2015-08-10:
Oh, sorry! I am using the latest version (downloaded half an hour before I opened this issue).
I click "Show In Progress Downloads" and there are two buttons. The first one does nothing (but I think it should open the correct Finder window) and the "X" button I haven't touched.
Right clicking doesn't do anything.

nwinkler, 2015-08-17:
I just ran into the same issue. The trick is to right-click on the list of distributions on the left side of the window, not in the _Progress_ window.

Right-clicking in the left side of the window shows the _View in Finder_ option, which you can use to open the downloaded distribution in Finder.

[image]

For me, it was downloaded to `/Users/My.User/Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/Data/Library/Application Support/Mac Linux USB Loader/Downloads`

ThePixelPony, 2015-08-18:
Thank you! I've found mine in `Application Support` too.

closed 2015-08-18 by ThePixelPony

karlic, 2016-05-13:
@SevenBits I don't think this behaviour is intuitive. I used MLUL to download Ubuntu. It disappeared from "View In-Progress Downloads" and when I selected "Create Live USB" the file finder window that opened showed my Documents folder. It didn't occur to me to right-click on the Distribution in MLUL and I had to search for it in Terminal.

I think Create Live USB should open the folder that contains the downloaded distributions.

SevenBits, 2016-05-14:
@karlic Perhaps distributions downloaded with the downloader should remain in the popover until the user quits the app?

Unfortunately, you can tell the open panel to show a directory by default, but I've found that it doesn't necessarily respect that depending on what settings the user has chosen. You can always create a USB with a downloaded ISO by double-clicking on the distribution name in the side panel.

karlic, 2016-05-14:
@SevenBits My preference would be to just download them to ~/Downloads/. For me that is the logical place to look for them. They are also big, so I don't want them "hidden" in ~/Library... and once I'd installed them to USB, I deleted them. (If I need them again, I'll download the latest ISO.)

I wouldn't have thought of double-clicking the distribution to install. Maybe dim of me, but that's the users you're working with. ;-)

Suggestions (in preference order):
1. Just download to ~/Downloads/.
2. After download completes change "Download This Distro" button to say "Install This Distro". (Maybe also change text colour to green to highlight the change.) Leave the double- and right-click functionality for clever people and backwards compatibility.
3. A panel in the Distribution Download window explaining that you can double click to install and right-click to view in Finder.
4. After download completes have an alert say, "Successfully downloaded to ~/Library..." (But the path is probably too long to remember for most.)
5. Prompt user for download location.