SevenBits/Mac-Linux-USB-Loader #82: No Kernal Found

ThePixelPony, 2015-08-15:
I've downloaded and loaded the Mint ISO under the option "Ubuntu" and selected both tick boxes.

When I boot holding down alt, I get the "EFI Boot" option, which I selected, following by booting "Ubuntu".
It said that it couldn't find a kernel. It tried loading a ramdisk but got an error that a kernel is needed first.
Next, I got the normal Linux Mint startup screen, so I selected the first option and the screen was frozen black for over 20 minutes. I powered off my mac.

SevenBits, 2015-08-15:
@ThePixelPony Which version of Mac Linux USB Loader are you using?

ThePixelPony, 2015-08-15:
@SevenBits, it's 3.1 (I believe the latest)

bduisenov, 2015-08-20:
have the same issue

SevenBits, 2015-08-21:
@ThePixelPony @bduisenov Please post the Enterprise configuration file. It is named `enterprise.cfg` and is located in the same folder as the ISO file and EFI system files on your USB.

bduisenov, 2015-08-21:
I fixed the issue using i386 iso instead of amd64+mac

SevenBits, 2015-08-21:
@bduisenov Okay, glad you got it working.

ThePixelPony, 2015-08-22:
@SevenBits, I shall paste it below.
>#This file is machine generated. Do not modify it unless you know what you are doing.
>
>entry Ubuntu
>family Ubuntu
>kernel /casper/vmlinuz file=/cdrom/preseed/ubuntu.seed
Please note that none of these locations mentioned in this file exist. The only folder in my USB stick is called "efi" (except the OS X generated ones).

SevenBits, 2015-08-22:
@ThePixelPony The locations mentioned in the file should exist inside of the ISO file. Can you mount your ISO file and check the paths inside of it?

ThePixelPony, 2015-08-22:
@SevenBits I have tried to mount the boot.iso but got the response "No mountable file systems".  This response is identical to the one from the original Linux Mint iso (and a Deepin iso that I have for that matter).

SevenBits, 2015-08-22:
@ThePixelPony Well, that's to be expected, although I can usually mount Ubuntu ISOs on my computer... Any chance you can use third-party software?

ThePixelPony, 2015-08-23:
@SevenBits, I am currently on holiday but I will mount it on my computer when I get back as I couldn't get hold of a third-party program. Sorry for all the trouble.

closed 2015-10-24 by SevenBits