SevenBits/Mac-Linux-USB-Loader #83: Kali 2.0: no controller found

kassi, 2015-09-01:
Following the guidelines with the latest kali-2.0-amd64.iso and adding a 128 MB persistent file to the same partition, I get the following error when booting from usb, after I chose option 1)

    [    1.899668] i8042: No controller found
    Loading, please wait...
    [    2.175150] usb usb2-port1: couldn't allocate usb_device
    [   12.589508] usb 1-5: device descriptor read/all, error -110
    
    
    BusyBox v1.22.1 (Debian 1:1.22.0-9+deb8u1) built-in shell (ash)
    Enter 'help' for a list of built-in commands.
    
    /bin/sh: can't access tty: job control turned off
    (initransfs) Unable to find a medium containing a live file system
    modprobe: module ehci-orion not found in modules.dep

SevenBits, 2015-09-01:
Does the error not appear if you turn off booting with persistence?

kassi, 2015-09-01:
It has nothing to do with persistence.
In fact, when entering the advanced state and boot then it works. Just pressing 1 on the main screen doesn't. However there's another thing: when I enter (now taken from my mind) the last option under advanced, the boot loader hangs and doesn't react anymore.

Anyways, looks strange. I managed to boot going through advanced menu and just start, but a lot of things are not working afterwards: no wifi, no bluetooth, strange trackpad behaviour. I actually wanted to check if direct boot works better with a wifi-usb dongle than within VirtualBox where it very often suddenly just kills the whole network, but it's not worth it - you can't work with it.

SevenBits, 2015-09-01:
Hmm, that's weird. The WiFi issues, etc. probably are driver related... Which model MacBook are you using?

kassi, 2015-09-02:
MacBook Pro (Retina, 15-inch, Late 2013)
2,6 GHz Intel Core i7

SevenBits, 2015-09-10:
@kassi Have you tried enabling the `nomodeset`

SevenBits, 2016-01-12:
Closing this as no additional info has been posted.

closed 2016-01-12 by SevenBits

JustSeb, 2016-06-05:
Did anyone fixed this problem? cant find any solutions online..