SevenBits/Mac-Linux-USB-Loader #84: multiple distro on the same key

sanchezg06, 2015-09-12:
Hi there,
Thanks for your great software! I would love to be able to boot different distributions from the same key. e.g. ubuntu kali tails that could be selected from the boot menu. Maybe it 's already possible. If it's not it could be a great feature to add!

SevenBits, 2015-09-12:
This is possible, but not officially supported by Mac Linux USB Loader; you have to set it up manually. You can find instructions here: https://sevenbits.github.io/Enterprise/.

karlic, 2016-05-13:
Newer instructions now available here: https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Configure-booting-(one)-of-multiple-ISOs-(multiboot)

I couldn't get the ones at https://sevenbits.github.io/Enterprise/ to work.

karlic, 2016-05-13:
@SevenBits I feel that when selecting "Create Live USB" MLUL should check whether the USB has already been MLUL'ed (not to be confused with [mullered](http://www.urbandictionary.com/define.php?term=mullered)) if so, the user given the option to reinstall from scratch or **add** the selected distribution.

**Adding** would just be a case of copying the ISO to /efi/boot/ and creating an entry in enterprise.cfg. The "boot.iso" of "/efi/boot/boot.iso" in lines 106--112 in SBEnterpriseConfigurationWriter.m 

		} else if (family == SBDistributionKali) {
			[string appendString:@"kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali\n"];
		} else if (family == SBDistributionTails) {
			[string appendString:@"kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live config live-media=removable noprompt timezone=Etc/UTC block.events_dfl_poll_msecs=1000 splash nox11autologin module=Tails quiet splash\n"];
		} else if (family == SBDistributionDebian) {
			[string appendString:@"kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live config live-config quiet splash"];
		}

should be replaced with the name of the ISO file selected.

In fact, replacing "boot.iso" with, say, "tails-i386-2.3.iso" would seem to be good practice as it makes it explicit what exactly will be booted when browsing /efi/boot/ or reading enterprise.cfg.

SevenBits, 2016-05-14:
@karlic Yes, that's a good suggestion, but I'm a little short on time right now (I have other full-time responsibilities). If you'd like to make the changes and submit a patch or pull request, I'd happily accept it!

karlic, 2016-05-14:
@SevenBits I just noticed that "splash" appears twice on line 109 in SBEnterpriseConfigurationWriter.m

If I get time I'll try and do this. I don't have a developer account though, so not sure how far I'll get in testing.

SevenBits, 2016-05-14:
@karlic Hmm, I didn't catch that. It's harmless in any regard, though, so not an immediate issue.