SevenBits/Mac-Linux-USB-Loader #85: Unknown error sending read/write command, card status 0x900

markwylde, 2015-09-17:
Hey, just tried your Mac Linux USB Loader and it's the closest I've got yet to having a LiveCD work, but unfortunately not there just yet.

I have tried both Debian and Ubuntu and the disk appears in the alt boot screen as expected. I select the disk and then it asks me what I would like to do. I hit 1, and then select Ubuntu (or Debian originally).

The screen flashes and then the text goes really small. This all seems fine, then after about 10 seconds I get the following:

[image]

I am using a MacBook Pro (Retina, 15-inch, Early 2013) 2.7GHz, i7 and the OSX on there is all up to date, running Yosemite 10.10.5

I am using a 128GB SD card from Transcend (JetDrive Lite 130) so it's not really a USB drive, but I presume that should be okay, since it gets to the boot screen and all okay.

Anyway fingers crossed someone has come across this and has an answer.

SevenBits, 2015-09-17:
@markwylde Your SD card should work, but just to try ruling things out, could you try it on a standard USB drive? Based on the error messages it seems that Linux is having some difficulty reading the drive...

markwylde, 2015-09-21:
Yeah it works as expected on a USB device :(

SevenBits, 2015-09-22:
@markwylde Looks like Linux can't read from your SD card then. Maybe it's a driver issue. Regardless, it's an issue with Linux and not Mac Linux USB Loader.

closed 2015-09-22 by SevenBits

JasonJWilliamsNY, 2015-11-13:
I get a similar message. I have tried this with several different USB drives (8-16GB). It takes >1000 seconds to boot. Happy to provide more details - this is the first (nearly) working solution I've found. Thanks!
[image]

SevenBits, 2015-11-18:
@JasonJWilliamsNY Does it ever boot completely? If it takes that long, my guess is that it's an I/O problem.