SevenBits/Mac-Linux-USB-Loader #86: USB flash drive not seen

monikamatis, 2015-09-19:
I tried Cruzer Edge 16GB drive to create a bootable usb stick for Ubuntu based distribution. All works nicely except for an important issue - there is no drive to choose from, despite the fact that the stick is mounted and working properly on the system.
System is Yosemite 10.10.5 run on a late 2014 Mac Mini

SevenBits, 2015-09-19:
What do you mean by:

> there is no drive to choose from, despite the fact that the stick is mounted and working properly on the system

Do you mean the USB doesn't show up in the boot manager?

monikamatis, 2015-09-19:
yes, exactly

SevenBits, 2015-09-21:
@monikamatis Can you please share the layout of files on your USB drive?

SevenBits, 2015-10-24:
Closing as this is a duplicate of #29.

closed 2015-10-24 by SevenBits