SevenBits/Mac-Linux-USB-Loader #87: Can't create USB loader in El Capitan

rbrown256, 2015-10-03:
Steps:

1. Insert USB stick into MacBook Pro (16GB Kingston).
2. Launch Mac Linux USB Loader.
3. Double click Create Live USB.
4. Select file "ubuntu-14.04.3-desktop-amd64.iso", 1.05 GB.
5. Choose KINGSTON as destination and leave dropdown as "Included With Application".
6. Leave Distribution Family as Ubuntu and leave options as "This ISO has code older than Ubuntu 14.10" checked and "This ISO lacks an EFI-enabled kernel" unchecked.
7. Click Begin Installation.
8. Error message is displayed "Couldn't get security scoped bookmarks."

[image]

SevenBits, 2015-10-04:
Try deleting `~/Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/Library/Application Support/Mac Linux USB Loader/EnterpriseInstallationLocations.plist`.

aschereT, 2015-10-04:
Probably that damned SIP. Try disabling it?

SevenBits, 2015-10-04:
@aschereT Mac Linux USB Loader doesn't write to any protected system locations so SIP is very likely not the issue.

afram, 2015-10-05:
@SevenBits same issue here, macbook pro mid 2015 15" el capitan. That location you suggested to delete doesn't exist on my machine.

CoreFloDev, 2015-10-05:
I just got the same issue, I closed the application, deleted this folder ``~/Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/`` with the command ``rm -R ~/Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/`` and it fixed the issue.

afram, 2015-10-05:
@CoreFloDev did it make it passed "Grant Access" for this drive?

After deleting that directory and its contents, I get the next screen, but after granting access, I get the same error shown above.

SevenBits, 2015-10-08:
@afram Try clearing the Mac Linux USB Loader USB cache with `defaults delete com.sevenbits.Mac-Linux-USB-Loader`.

hpnishi, 2015-10-08:
@SevenBits I got the exactly same problem as @afram.
After granting access I receive the same error. 
I already deleted com.sevenbits.Mac-Linux-USB-Loader, but nothing changed.

hpnishi, 2015-10-08:
I found a way to solve it.
1 - downloaded the previous version (3.0.2)
2 - tried to create the bootable USB, but since Kali didn't appear as an option in the Distribution Family I gave up and deleted the old version.
3 - after that, suddenly, the new version (3.1) worked like a charm.

I just don't know if that would work for everyone.

paulopaniago, 2015-10-15:
I’m also have this problem, and I tried on both: Yosemite and El Capitan. I tried also de above suggestions and none of them worked with me. This is weird because I had already made a linux stick before using Mac Usb Loader on Yosemite. (although it didn’t worked in the end because the ISO I got in the net was incompatible, I guess)

paulopaniago, 2015-10-15:
hello, just solved this problem. I downloaded the version for lion (2.0.1). I can′t believe I have a pendrive with linux now. Thanks very much for the loader SevenBits!!!

paulopaniago, 2015-10-15:
Hello SevenBits, my Linux live usb is working fine after I used version 2.0.1 of Mac Linux Usb Loader. But unfortunately the data is all lost after reboot. Do you know how can I add Persistence app to that usb stick now?
I am thinking about migrating to Linux because Apple prices in Brazil became way overpowering for me to continue with MacBooks. I intend to use your program′s created usb combined with another open source program to simulate a stenomachine – very important to me. In case you’re curious, I’ve opened a topic about this treat in their forum:
https://groups.google.com/forum/#!topic/ploversteno/7KLi7ssQd94

SevenBits, 2015-10-16:
@paulopaniago I'm curious about this, as I haven't had any of these issues. I would recommend against going down to 2.0.1, though. It is running an old and outdated version of Enterprise and doesn't support the newer features of Mac Linux USB Loader.

Apple probably changed some code in El Capitan relating to security bookmarks, which I use to co-ordinate access to USB drives. Does it still work on Yosemite? I would recommend against anyone upgrading to El Capitan until I get to the bottom of this.

paulopaniago, 2015-10-16:
Hello SevenBits thanks so much for answering.
And yes I’m having the same problem on both systems: El Capitan and Yosemite.
But I have used your program before in Yosemite and it worked fine. It didn’t after I upgraded it to El Capitan. Because of other incompatibilities I decided to reinstall Yosemite, but this time I’ve separated my internal driver in two partitions, one for each system. 
Unfortunately now I have the same problem on Yosemite and on El Capitan. Guess they changed something in Yosemite’s update.
Someone told me a pendrive would be slow with Linux, but now I’m thinking about buying an external usb 3.0 SSD of about 60gb. I guess with that I would have the best of both worlds: a portable drive and a fast operational system.

SevenBits, 2015-10-17:
@paulopaniago Can you email me a copy of the Mac Linux USB Loader and system logs? I will contact Apple about this problem.

FracturedLoop, 2015-10-27:
I tried what SevenBits said and it worked: 

>Try clearing the Mac Linux USB Loader USB cache with defaults delete com.sevenbits.Mac-Linux-USB-Loader.

kapitainsky, 2015-11-03:
deleting com.sevenbits.Mac-Linux-USB-Loader works for me as well. Small inconvenience as have to do this every time I want to create USB. BTW - I am on El Capitan.

SevenBits, 2015-11-03:
@kapitainsky Something must have changed in El Capitan, as everything works as expected on Yosemite. I'll work on resolving this issue ASAP.

Could you try disabling System Integrity Protection and seeing if that helps the issue at all?

kapitainsky, 2015-11-04:
I have disabled SIP - after restart created successfully Ubuntu USB. However when I have tried to create another one, had the same issue (screenshot below) immediately after clicking 'Begin installation' button:

[image]

Removing app (used AppCleaner - files removed detailed below) hasn't solved anything. Restarted computer and then could create one USB again. It behaves pretty much the same as with SIP enabled. Sometimes I can have 'Grant access' dialog and error after sometime before... So far have not noticed clear pattern.

[image]

maxious, 2015-11-07:
Same outcome on 10.11.1. Commands removing cache/temp files don't help but restarting does.

SevenBits, 2015-11-07:
@maxious @kapitainsky The only thing that seems to help, other than restarting, is clearing Mac Linux USB Loader's defaults:

    defaults delete com.sevenbits.Mac-Linux-USB-Loader

I am investigating these issues and should have a fix soon.

ksasa001, 2015-11-17:
I created a bootable USB previously on El Capitan, but tried recently and got the error of, "couldn't get security scoped bookmark." The solution above worked for me.

`defaults delete com.sevenbits.Mac-Linux-USB-Loader`

SevenBits, 2015-12-09:
This issue should be resolved somewhat in version 3.2.

akafester, 2015-12-22:
Still happening on 3.2. Tried with other USB drives but the only thing that resolved it was the solution above: defaults delete com.sevenbits.Mac-Linux-USB-Loader

SevenBits, 2015-12-22:
You shouldn't need to delete the defaults. Simply open Mac Linux USB Loader, go under the application menu (i.e the one labeled Mac Linux USB Loader), and under the Registered Devices menu, click on your USB drive to remove it from memory. Then, you will be asked for authorization again and everything will work.

This error occurs because when you re-format the USB drive with the same name as before, the file system has a different ID, so while Mac Linux USB Loader knows it's the same drive, OS X believes it is different, and so denies access. The easiest way to avoid this problem is either to not format the USB drive before you use it with Mac Linux USB Loader (it's not necessary, despite what some online tutorials say), or if you must re-format, don't use the same name as before.

stefankoch, 2015-12-24:
Hi in my case I tried everything above. Deleting the suggested folder in the library, emptying the cache, emptying the defaults, etc etc. The only thing that worked is creating an image with another name as suggested in the last reply by SevenBits :) Thanks guys - now let's see if it boots with Ubuntu Rescue Remix 12.04

stefankoch, 2015-12-24:
sorry to clarify I mean when formatting the USB drive in disk utility give the 1 partition fat and mbr setup a new name than before, otherwise bookmarks are confused. This could improved or at least hinted at in the dialogs in 3.3 -- thanks & happy holidays! 
https://developer.apple.com/library/mac/documentation/Security/Conceptual/AppSandboxDesignGuide/AppSandboxInDepth/AppSandboxInDepth.html#//apple_ref/doc/uid/TP40011183-CH3-SW16

stefankoch, 2015-12-24:
now I can run the installer, but the boot stick doesnt run with Ubuntu Rescue Remix even if both options (for versions below 14.x and doesnt include EFI) are checked. First error 'file /casper/vmlinuz.efi not found' then kernel panic 'unable to mount root fs on unknown block'.. any help would be appreciated thanks

SevenBits, 2015-12-24:
@stefankoch I'm not familiar with Ubuntu Rescue Remix. I will try to test it.

stefankoch, 2015-12-25:
Here is the link to the ISO file download, to reproduce simply load this with Linux Mac USB Loader on a fat-formatted stick and try booting the mac (OSX 10.9 or 10.10) holding 'alt':
http://iso.linuxquestions.org/ubuntu-rescue-remix/ubuntu-rescue-remix-12.04/#x86
I get the error '/casper/initrd.lz' not found, which is probably because it is in gz format (rather than lz) in this live distro:
The vmlinuz.efi error message may be solvable by simply renaming the file, but modifying isos is a painful process on macs..
https://wiki.ubuntu.com/CustomizeLiveInitrd

SevenBits, 2015-12-27:
@stefankoch I downloaded Ubuntu Rescue Remix and it worked just fine for me once I put in the necessary settings. To reproduce:

1. Open Mac Linux USB Loader and select the Ubuntu Rescue Remix ISO file
2. Select your desired USB drive on the first screen
3. On the second screen, make sure the distribution family is set to be "Ubuntu" and check __both__ checkboxes (Ubuntu RE has a package base older than Ubuntu 14.04 _and_ its kernel isn't EFI enabled -- you know this because its name doesn't end with a .efi extension)
4. Hit the Begin Installation button
5. Out of the box, Mac Linux USB Loader thinks that your ISO has an initrd named `initrd.lz`, but in the case it's `initrd.gz`. To account for this, when the installation is done, open the Setup USB Device screen from the main menu and select the option to edit the Enterprise configuration file on your USB
6. At the bottom of the file, add the line:

> initrd /casper/initrd.gz

stefankoch, 2015-12-27:
Hi thanks that got me further. However, after Ubuntu begins booting
slightly, I get the message
'could not find the ISO /efi/boot/boot.iso'
Could be another similar path question but not sure - what do you think?

SevenBits, 2015-12-27:
@stefankoch I was able to get it to work using the exact steps I posted above. Maybe post your configuration file?

stefankoch, 2015-12-27:
I just added the line you mentioned with both options checked (i.e. no
EFI..) and OS recognized as Ubuntu. It boots into the command line using my
2014 Macbook Pro but not using the 2012 11 Inch Macbook Air (MBA) - the one
where it is needed. Does that help?

stefankoch, 2015-12-27:
here is the configuration file I'm using:

#This file is machine generated. Do not modify it unless you know what you
are doing.

entry Ubuntu
family Ubuntu
kernel /casper/vmlinuz file=/cdrom/preseed/ubuntu.seed
initrd /casper/initrd.gz

SevenBits, 2015-12-27:
@stefankoch Maybe try booting with the `nomodeset` option enabled?

SevenBits, 2016-01-12:
A workaround for this issue was added in version 3.2, so I’m closing this issue now.

closed 2016-01-12 by SevenBits

gti3993, 2016-03-01:
Well Before we get all complicated and geeky ,When you see the Grant Access make sure you select the usb drive not the iso or anything else ! it just needs to sudo over usb drive !

alkibia, 2016-08-13:
works perfect for me: 
> defaults delete com.sevenbits.Mac-Linux-USB-Loader.

SevenBits, 2016-08-14:
@alkibia and everyone else: this issue should be resolved now in the latest releases. Just please keep in mind @gti3993’s comment:

> Well, I guess before we get all complicated and geeky, the easy fix is, when you see the Grant Access pop-up screen, just to select the usb drive -- not the iso or anything else! it just needs to sudo over usb drive! so Grant Access to usb and move on!

JonDotKom, 2016-08-30:
defaults delete com.sevenbits.Mac-Linux-USB-Loader: Works

sudo rm-rf  ~/Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/ : Works

The next time it prompted me to grant the access & still got the issue: I changed the name to disk drive... Perfect

baileyturner22, 2016-09-02:
Hey guys i've been having the same issue with Mac linux usb loader, I have tried all of these fixes that you have suggested on here and none of them have worked. So I began messing around doing random things and fixed it.< GO TO THE TOP LEFT CORNER AND CLICK ON MAC LINUX USB LOADER  NEXT TO THE APPLE LOGO, GO DOWN TO REGISTERED DEVICES AND CLICK ON THE USB NAME OR CLEAR ALL.> This worked for me sorry if it doesn't for you good luck.

SevenBits, 2016-09-02:
@baileyturner22 Yes, this solution is detailed [in the FAQs](https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Frequently-Asked-Questions#4b-i-get-a-message-saying-the-usb-device-that-you-have-selected-cannot-be-accessed-because-the-system-denied-access-to-the-resource).

JonDotKom, 2016-09-03:
Yes.. I agree.. I just tried the solutions above before @baileyturner22 & it worked for me this time but there has been times it has not.. I have went to the drop down before but I forgot about it & it works everytime

tbird20d, 2016-09-30:
Problem solved!  I had the same problem.  My setup was: Macbook pro retina, El Capitan, 10.11.4.  Sandisk Cruzer 8GB, with MSDOS (FAT32) filesystem, ubuntu-16.04.01-desktop-amd64.iso, and Mac Linux USB Loader version 3.3.

I tried deleting the directory, doing 'defaults delete', and clearing the registered devices in the Mac Linux USB Loader menus.  These didn't work (or, they didn't work alone).  It worked when I followed the suggestion of @gti3993.  When the dialog appears requesting that you Grant Access, go to the left pane and select the USB device, then click on "Grant Access".

When I re-read the General instructions page, the instruction to make sure the "Grant Access" dialog is at the root of the USB drive is right there. Doh!

I'm putting this here in case anyone else failed to understand, or misread the instructions, like I did.

jamesalmeida, 2018-10-31:
> Well i guess before we get all complicated and geeky ,the easy fix for When you see the Grant Access PopUp Screen is just to select the usb drive not the iso or anything else ! it just needs to sudo over usb drive ! so Grant Access to usb and move on !

Sweet. This worked for me. Thanks!