SevenBits/Mac-Linux-USB-Loader #88: Why is not in English

wisdomlight, 2015-10-03:
Hi,
I've downloaded and opened the app.

I've downloaded it three times.

it's in German

Is there an English version?

Cheers

SevenBits, 2015-10-04:
Mac Linux USB Loader appears in whatever language your system is set to. Change your system language to English if you want the app to be in English.

tveloq, 2015-10-09:
I had the same issue although my system language was set to English.

In order to fix it I had to browse the app with "Show package contents" and delete the following folders:

/Contents/Resources/de.lproj
/Contents/Resources/es.lproj
/Contents/Resources/nl.lproj

SevenBits, 2015-10-10:
@tveloq That's very strange. I'm not sure what could be causing it.

closed 2016-01-12 by SevenBits

borgesrodrigo, 2016-02-12:
i'm having the same issue.

eMerzh, 2016-04-13:
Same here.. last version. the app is in NL and the system is in French...
i think the issue must be re-opened

SevenBits, 2016-04-13:
@eMerzh Unfortunately, without a way to reproduce the problem on my end, it’s going to be very difficult to fix the problem, but fell free to re-open if the issue still exists.

eMerzh, 2016-04-14:
@SevenBits OK... But thé issue is still there in thé last version. Juste donwloaded it, unzip then run. And i've not the ability to re open the issue.

It might be the code that determine the user's language or smth? Is there some way we can give you more info to put you on track?