SevenBits/Mac-Linux-USB-Loader #89: Can't set persistence!

hpnishi, 2015-10-09:
I finally figured out how to create the live USB boot for Kali, but now I can't set persistence.
Mac Linux USB Loader doesn't show any pendrive:
[image]

Best regards,

SevenBits, 2015-10-09:
In order for persistence to work with Kali, it must meet special requirements for security reasons. I believe you need to partition your USB drive. If you search on Google you can get more detailed information.

closed 2015-11-03 by SevenBits

bnhorowitz, 2019-05-28:
I just installed Ubuntu 18.04 on a USB and have the same problem as above. Mac Linux USB Loader does not recognize the USB drive to set persistence.