SevenBits/Mac-Linux-USB-Loader #90: Slider does not show space for Persistance

AcexSpade, 2015-10-21:
Slider does not show space for Persistance on El Capitan with version 3.1

SevenBits, 2015-10-21:
Can you be more specific? Perhaps include a screen shot?

rohannarang15, 2015-10-29:
Setting the slider doesn't seem to update the value in the text box. It stays at 128 mb
[image]
I went ahead and put the slider to its max value and pressed "Create Persistence" anyway. After the operation was done, I checked the size of the casper-rw file and it looks like the operation was successful.
[image]
So I think this is just a minor GUI bug. I'm using El Capitan.

nnevvinn, 2015-10-31:
I had the same experience as rohannarang15 on 10.11.1 -- no response to moving slider, but created an ≈4gb file.

SevenBits, 2015-11-03:
@nnevvinn @rohannarang15 This is a known issue. I will put out a fix ASAP.

keyvanfatehi, 2015-11-07:
I have the same issue. Running OS X Yosemite

SevenBits, 2015-11-27:
This issue has been fixed and the fix will be released within the next few days.

closed 2015-11-27 by SevenBits