SevenBits/Mac-Linux-USB-Loader #94: where is the file directory? v3.0.2

housingfreq, 2015-11-22:
Hi,

I'm using the version 3.0.2, downloaded from your home page, can't use the latest since its in german or something.

I downloaded it twice, when I click the finder icon, it didn't appear or pop out, so I want to manually go locate the 14.10 ubuntu distro i downloaded. 

Where is the distro downloaded/saved to?

Thanks.

SevenBits, 2015-11-22:
@housingfreq I have received a number of reports about the application showing up in German. What language is your system set to?

housingfreq, 2015-11-22:
US English

May i know where is the saved location of the distro? I want to manually go and remove, if not its eating my HDD space.

SevenBits, 2015-11-22:
@housingfreq If you right click on the distribution name in the downloader, there should be an option to "View in Finder". Select that and it will take you to the file.

SevenBits, 2015-11-23:
@housingfreq Also, please show me your language setup. It looks like this:

[image]

housingfreq, 2015-11-23:
the image:
https://photos-4.dropbox.com/t/2/AACTW37VnXa8vTwcvoPVsLZXzL4vDWxt02ErvFfuIM6aAQ/12/44430038/png/32x32/1/_/1/2/Screen%20Shot%202015-11-23%20at%208.31.58%20AM.png/EJWFgSIYlJoOIAIoAg/mEmVvGNbjadH2fqSP30oNsy08KG0X-TH64lajkInqGg?size=1024x768&size_mode=3

For that method, previously I have tried, it didn't work, I clicked once or double clicked it, it didn't open anything. 

Could you please let me know the saved distro iso image location? I wish to manually go in and remove those saved images. Thanks.

SevenBits, 2015-11-23:
The link you gave doesn't work; it gives me an error 403.

Off the top of my head, it should be something like `Library/Containers/com.sevenbits.Mac-Linux-USB-Loader/Library/Application Support/Mac Linux USB Loader/Downloads/`.

housingfreq, 2015-11-23:
I'm sorry, i have edited the image link again, it should be working now. 
Perhaps a button your application to adjust language, that will temporary solve the issues for future users.

I solved the language problem by opened the folder of the application and deleted those language folders, followed solution method suggested by the other user.

SevenBits, 2015-11-25:
@housingfreq Were you able to find the ISO at the path provided? If so, please close this issue.

housingfreq, 2015-11-26:
okay thanks, i will close it.

closed 2015-11-26 by housingfreq