SevenBits/Mac-Linux-USB-Loader #95: Disk Space on Kali only 970MB

goodguinness, 2015-11-25:
Hi,

Problem: Kali boots fine, however when I view my drive on Kali, it only has about 970 MB for disk storage even though my USB drive has nearly 60 GB of free space. I cannot update Kali or do anything that takes up storage because of this.

Info: I am running Kali Linux on my MacBook Air using Mac Linux USB Loader. I have formatted the USB drive to FAT and set up persistence. Kali is bootable but only able to use 970 MB of storage. I am hoping to have the OS recognize the USB drive so I can save files and update Kali. Please help!

Thank you.

goodguinness, 2015-11-25:
Mac Linux USB Loader 3.1
Kali Linux iso downloaded through the distribution downloader
MacBook Air / OSX Yosemite 
SanDisk 64 GB drive

goodguinness, 2015-11-27:
Also, persistence is not working. Thanks.

SevenBits, 2015-11-28:
Mac Linux USB Loader cannot set up persistence that can be used by Kali, since Kali requires its own dedicated partition. Ubuntu is much better in this respect.

closed 2015-11-28 by SevenBits