SevenBits/Mac-Linux-USB-Loader #97: Installation Options needs explanation

SeanFelipe, 2015-12-08:
At the final screen, a couple options appear which, to me at least, are a bit obscure. 'This ISO lacks an EFI-enabled kernel' and 'This ISO has code older that Ubuntu 14.10'.

Not being intimately familiar with these aspects of Linux -- the Ubuntu versioning system is straightforward I suppose, but pretty annoying to me -- I really have no idea what to select.

Can we have some explanation of what these mean, how to find out the answer, and why it's important?Without explanation, it just creates uncertainty for those who are not up to speed with the technology.

Thanks for the tool - I donated!

SevenBits, 2015-12-09:
@SeanFelipe I'll do a full write-up tomorrow. Thanks for sharing your feedback!

SevenBits, 2016-06-12:
I’m working on improving this now.

closed 2017-04-28 by SevenBits