SevenBits/Mac-Linux-USB-Loader #98: Many Problems Attempt to Create USB-bootable Drive for Ubuntu-Server

planewryter, 2015-12-12:
Hi Seven Bits,

Please see the attached PDF with explanation and screenshots of a series of problems encountered attempting to use Mac Linux USB Loader to create a bootable USB-based instance of Ubuntu Server.

Thank you very, very much for any assistance!

[Seven-Bits-Request.pages.pdf](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/60222/Seven-Bits-Request.pages.pdf)

SevenBits, 2015-12-12:
@planewryter Well, you've won the award for the most detail ever put into an issue report. Congratulations!

Despite your confusion, the problem is actually quite simple. Mac Linux USB Loader is setup to install Ubuntu desktop, not Ubuntu server. You can still use the server version, though, but it requires different settings. What settings did you use in Mac Linux USB Loader to create the flash drive?

To answer your question, Enterprise is a software tool installed onto your flash drive so your Mac can boot Linux. It comes bundled with Mac Linux USB Loader.

planewryter, 2015-12-13:
Hi Seven Bits!

Your assistance is greatly valued & much appreciated!

Hope you’ll be so kind as to guide me to the Land of Milk and Ubuntu Server booting from a USB Drive on my iMac and/or MacBook Air.

When you have a moment, please see the attached PDF (yea! More screenshots!).

Many, many thanks!

Best,
Rick

SevenBits, 2015-12-13:
@planewryter Hi there Rick. There is no attachment here on GitHub. I think you need to upload it from the site, not from email.

In Mac Linux USB Loader, open the USB Setup window (located on the main screen), and then select your USB drive and click `Edit Enterprise Configuration File`. TextEdit will open and there should be some text in the window. Paste the text here on GitHub. Once I see that, I can give you some better advice.

planewryter, 2015-12-13:
[Seven-Bits-Request-Too.pages.pdf](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/60610/Seven-Bits-Request-Too.pages.pdf)

Edit Enterprise Configuration File does not seem to function. Is there a path I can follow in TextEdit to open the affected Configuration File?

Thanks!

SevenBits, 2015-12-13:
@planewryter The first "surprise" that you ran into is unfortunately a drawback with OS X El Capitan that I can't get around due to system restrictions. Luckily for you, Mac Linux USB Loader's documentation includes [a workaround](https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Frequently-Asked-Questions#4b-i-get-a-message-saying-the-usb-device-that-you-have-selected-cannot-be-accessed-because-the-system-denied-access-to-the-resource) that you can use. The best way to avoid this issue is not to format your drive before you are going to use it. Instead, go to your drive and delete its EFI folder (I'll be adding this option to Mac Linux USB Loader).

Next time you create a drive with Ubuntu Server on it, please check the "this ISO lacks an EFI-enabled kernel option". Ubuntu desktop includes support for EFI, but I believe Ubuntu server does not. If you don't know what this means, don't worry.

You said that the Edit Enterprise Configuration File button didn't work. Can you please describe the error for me?

Also, your PDF contains a lot of valuable feedback. I consider it valuable and I will be working to make the interface easier to understand for less-technical users. Thank you.

planewryter, 2015-12-13:
Seven Bits,

Thanks for the note about the workaround. I did check the FAQs earlier in the process (first attempt) and didn't think to access them on the second attempt. (I, of all people know...when all else fails: RTM.)

After De-Registering my USB drive...then what? Start over using the "...ISO lacks an EFI..." option?

As for "Edit Enterprise Configuration File," please see embedded video**. You will see a series of clicks, each followed by....{nothing, nada, zilch, zero, null--you get the idea}.

**UPDATE: Apparently, a video cannot be embedded in this comment system; trying to insert a URL to a Dropbox link; here: https://www.dropbox.com/s/t3gkr7pkt344e0f/Seven-Bits-3-010.mp4?dl=0. The video is about 12 seconds long.

Many thanks!

planewryter, 2015-12-13:
Seven Bits,

What seemed to be progress...

[image]

Alas--didn't yield much. Please see another link to a Dropbox video...this one showing attempted boot of Ubuntu Server from USB drive; here: https://www.dropbox.com/s/h5irnzzmxtghwo6/2015-12-13%2015.34.01.mov?dl=0

Your thoughts?

THANKS!

SevenBits, 2015-12-14:
@planewryter 
> As for "Edit Enterprise Configuration File," please see embedded video**. You will see a series of clicks, each followed by....{nothing, nada, zilch, zero, null--you get the idea}.

You need to do the `Edit Enterprise Configuration File` option **after** you create the USB drive with Mac Linux USB Loader.

> After De-Registering my USB drive...then what? Start over using the "...ISO lacks an EFI..." option?

Yes.

I'll download Ubuntu server and have a look at this. I've never had anyone try to do this before...

planewryter, 2015-12-14:
Here is the contents of the Edit Enterprise Configuration File:

#This file is machine generated. Do not modify it unless you know what you are doing.

entry Ubuntu
family Ubuntu
kernel /casper/vmlinuz

SevenBits, 2015-12-14:
@planewryter Thank you.

planewryter, 2015-12-18:
Hi SevenBits,

Because my team is having a meeting tomorrow morning…and the subject of Linux Server Admin is going to be at the top of the list—please accept this Email as a “check-in.” By any chance, were you able to provision Ubuntu Server on a Mac via Mac Linux USB Loader?

Many thanks for all!

Best,
Rick

SevenBits, 2015-12-18:
@planewryter Unfortunately, I don't have access to my Mac right now so I haven't been able to test. At some point tomorrow I'll tell you what I've figured out.

stefankoch, 2015-12-24:
Hi I have a similar issue using OSX Yosemite (10.10.5) using Mac Linux USB loader 3.2. The first time I used the app on my Stick it worked and I could 'authorize' it to access the stick and create an image, SystemRescueCD (a version of Gentoo). This would only boot to grub2 and then complained that the path to the kernel could not be found and therefore probably needs a special enterprise.cfg file. 

Since then when I run the app Mac Linux USB loader, and try to create, it is stuck in a loop asking me to grant access to the dir. If I hit cancel I get the error message "coudn't get security scoped bookmarks" (same as screenshot in the pdf above). 
To fix this, I tried 'registered devices -> clear all' from the app's menu. I also tried 'preferences -> clear caches now'. Finally I tried going into the Application Support folder and to delete a plist file and data folder for Mac Linux USB loader.
The issue may have to due with a missing 'entitlement' but no idea what that is:
http://stackoverflow.com/questions/10043345/trouble-creating-security-scoped-bookmark

Any help would be much appreciated, since I'm trying to rescue data from my macbook air after deleting the partition info by mistake. SystemRescueCD or Ubuntu are needed for this.
Best, Andrew

stefankoch, 2015-12-24:
ps I thought about using a second stick as a workaround since it seems to work the first time. I will try and post any results or non-results.

stefankoch, 2015-12-24:
update: unfortunately, using a different usb stick (formatted as Fat with MBR using Disk Utility) did not solve the problem still get the error msg 'couldn't get scoped security bookmarks'.

closed 2016-05-14 by SevenBits