SevenBits/Mac-Linux-USB-Loader #100: crash when selected "Accept HFS+ formatted Drives"

wukakuki, 2016-01-11:
I'm using OS 10.11.3 Beta (15D13b), mac linux usb loader 3.2(9345),kali 2.0.
It crash after I click the "create live usb" and choose the "kali-linux-amd64.iso" file , with "Accept HFS+ formatted drives" selected.
[“Mac Linux USB Loader” Sample.txt](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/86337/Mac.Linux.USB.Loader.Sample.txt)

SevenBits, 2016-01-12:
@wukakuki Hi there. Can I see this actual crash report as opposed to the process sample from Activity Monitor?

wukakuki, 2016-01-12:
@SevenBits 
Oh! After updated to OS 10.11.4 Beta (15E27e), this problem fixed. I'm try to boot my mac from USB  drive now.But my usb drive didn't show up. And I had tried to set my usb drive as startup disk. It failed with "Failure to establish boot caches on boot assistant partition".How to fix it?

SevenBits, 2016-01-13:
@wukakuki I’ve never seen this issue before. Can I see a screenshot of that error message and a copy of the system logs?

wukakuki, 2016-01-14:
@SevenBits 
It crash again. Can't select HFS+ again.
That screenshot as .gif and the error log in uname.pdf
And my system logs as system.pdf(too much logs, I don't know which one you need)

[system.pdf](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/90079/system.pdf)

[image]

[uname.pdf](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/90068/uname.pdf)

SevenBits, 2016-01-14:
@wukakuki Can you upload the screenshot as an actual video file? I can’t get enough details from that small gif...

wukakuki, 2016-01-16:
@SevenBits Oh! I'm sorry! I didn't notice that. However, it can't open any more.
I can't send you mov files via github.
when I try to send you .mov file, this message shows up.
and the crash log.
[log.pdf](https://github.com/SevenBits/Mac-Linux-USB-Loader/files/92695/log.pdf)

"Unfortunately, we don’t support that file type.  Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP."

SevenBits, 2016-01-16:
@wukakuki Send me the log files over email.

SevenBits, 2016-08-28:
Closing because there is no new information posted.

closed 2016-08-28 by SevenBits