SevenBits/Mac-Linux-USB-Loader #105: problem creating usb with Unbuntu

norton850, 2016-02-17:
Hello
I've tried creating an usb loader with my Macbook under 10.11.3 and Mac Linux USB loader.
I've want to use ubuntu 14.04.3 desktop-amd64.iso.
I can go on insatallation until I've got this message "Couldn't get security scoped bookmarks. The USB device that you have selceted cannot be accessed because the system denied access to the resource"
Can you help me ?
Many thanks
Philippe

SevenBits, 2016-02-18:
@norton850 RTM.

https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Frequently-Asked-Questions#4b-i-get-a-message-saying-the-usb-device-that-you-have-selected-cannot-be-accessed-because-the-system-denied-access-to-the-resource

closed 2016-02-18 by SevenBits

norton850, 2016-02-21:
Hello
many thanks for your reply.
I've still have problem to create USB drive with Ubuntu 14.0.
I use a usb drive with 4.0go.
When I launch the Mac USB loader program:
- I choose destination: the name of my USB drive
- go next
- distribution family: Ubuntu
- check this ISO has code older than Ubuntu 14;10
- begin installation
- "to authorize Mac Linux USB loader to acess your USB drive, please click Grant Acess below"
- I click "Grant acess
- I go back to the previous window with "to authorize Mac Linux USB loader...."
- I click cancel and have this message " Couldn't get security scoped bookmarks"
- so I go into Mac Lunix USB loader menu and select Registred device and clear all
- I tried the all process but still have the same problem???
Perhaps I do something wrong????
Many thanks for your help
Philipppe

SevenBits, 2016-02-21:
@norton850 Please go under the Application menu (the one titled "Mac Linux USB Loader") and go to the Registered Devices menu. Is your USB drive listed under the Registered Devices menu?

sboagibm, 2017-02-14:
@SevenBits I've read all of https://github.com/SevenBits/Mac-Linux-USB-Loader/issues/87 and your faq.  The USB disk has never once appeared under the Registered Devices menu.  I've tried a mountain of stuff, for a couple of hours, and that message still comes up.  Very frustrated!

SevenBits, 2017-02-14:
@sboagibm Please give me some more information. Maybe post a screenshot of the USB drive's information screen?

sboagibm, 2017-02-14:
@SevenBits here is the whole story in pictures.
[image]
[image]
[image]
[image]
[image]

I guess the OS is Yosemite, MacBook Pro (Retina, 13-inch, Mic 2014).

sboagibm, 2017-02-15:
@SevenBits any word on this? ^^^

SevenBits, 2017-02-15:
I notified you were presented with a Grant Access screen. Did it not work after that?

sboagibm, 2017-02-17:
> I notified you were presented with a Grant Access screen. Did it not work after that?

@SevenBits No, I'm still stuck.  I should try to explore more options for getting this done, but I payed for the software, so would be nice if it worked for my purposes (I'm following some directions which recommend using your product).

The screens I sent weren't in order, sorry.  When presented with "Grant Access screen", I get the "Couldn't get security scoped bookmarks" error box.

sboagibm, 2017-02-17:
@SevenBits I tried a different, 4gb ssd, same thing.  Given this has is a corporate installed system with disk encryption and firewall, I'm wondering if it could be some security thing that is unique to my install.

Ddfulton, 2017-07-22:
@SevenBits Same thing, with "Skip the boot selection menu" checked and with it unchecked, even after I click "grant access", it still stops me and says the system prevented it. Don't know how to fix it. It's too bad. Any headway on this issue?

SevenBits, 2017-07-23:
@Ddfulton You never saw the Grant Access screen shown above? And have you read [this page](https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Frequently-Asked-Questions#4b-i-get-a-message-saying-the-usb-device-that-you-have-selected-cannot-be-accessed-because-the-system-denied-access-to-the-resource)?

Ddfulton, 2017-07-23:
Yeah, I read the page. Cleared registered devices, system still doesn't grant access. Can't do anything else, either. Persistence Manager, Advanced USB Setup Options, they all put me in a loop of clicking "Grant Access." 

Maybe I could throw a sudo or two in to your source code? But where?

**Some more details**
OS X 10.12.5
Macbook Pro Retina 15 Inch Mid 2014

SevenBits, 2017-07-23:
Are you sure that you're clicking 'Grant Access' in the right place? When you click the button, the file viewer needs to be showing the root directory of the USB you wish to write to.