SevenBits/Mac-Linux-USB-Loader #106: Stuck on "Welcome to Enterprise!"

naerymdan, 2016-04-07:
**Short Description of Problem:** Super recent MacMini bought in April 2016 sees the usb key at boot and starts from it, stuck on "Welcome to Enterprise!" screen, nothing else happens.

**Expected Behavior:** 

Menu?

**Actual Behavior:** 

Noting.

**Steps to Reproduce:** 

- Using MacBookPro running ElCapitain (10.11), formatted 8GB key with FAT and GUID partition scheme.

- Installed ubuntu 15.10 AMD64 iso using the Loader

- Insert into MacMini, boot from usb

SevenBits, 2016-04-07:
Hi @naerymdan, thanks for posting. Can you please post your enterprise.cfg file (located in the /EFI/boot/ directory)? Thanks.

SevenBits, 2016-07-20:
Closing due to lack of activity.

closed 2016-07-20 by SevenBits