SevenBits/Mac-Linux-USB-Loader #107: After clicking on the live usb option and selecting the iso, nothing happens.

RikerW, 2016-04-08:
I click on the live usb option in the main menu thing.  The penguin with the `Create Live USB`.  

I selected the Linux Mint with cinnamon 64 bit iso.  I click open. 

Nothing happens.  The application is still open, but no new windows open.

I waited 15 minutes and still nothing happened.

I use OS X El Capitan (10.11) on a 2015 MacbookPro.

Clicking the icon in the dock just opens a new main menu, like at startup.

SevenBits, 2016-04-08:
What version of OS X are you running?

RikerW, 2016-04-08:
10.11

RikerW, 2016-04-08:
And I think I found my problem, somehow a period got put after the filename, so it was `foo.iso.`, I think it rejected it as a invalid file.

SevenBits, 2016-04-08:
That's probably it -- remove the period and see if it works.

RikerW, 2016-04-09:
Did, thanks.

closed 2016-04-09 by RikerW