SevenBits/Mac-Linux-USB-Loader #108: Booting Kali gets stuck before the GUI loads

rubert34, 2016-04-09:
I'm trying to run Kali from a USB drive, but it doesn't boot. Not sure what to do.

Mac Linux USB Loader seems to install everything ok.
When I restart and hold "alt", I'm presented with the choice to boot Kali, which I do.
After that it starts to load, but it gets stuck or never successfully complete.
There are different cases, depending on the procedure.

In normal mode, I usually get "kali gnu/linux comes with absolutely no warranty, to the extent permitted by applicable law" then it gets stuck.
The GUI loaded up one time, by using other options (don't remember which ones), but the keyboard and touchpad wouldn't respond.

I guess I must be missing something, but I'm not sure what. Any idea?

SevenBits, 2016-04-09:
Try booting with `nomodeset` enabled in the Advanced Options menu.

rubert34, 2016-04-09:
Not working. It seems to start loading something, but then gets stuck. There is a white dash in the top left corner and I can't type anything.

rubert34, 2016-04-09:
I made a video of this. Here it is: https://www.youtube.com/watch?v=TVEX7MIgfr8&feature=youtu.be
Now it does the same whether I boot normally or with the nomodset option.

SevenBits, 2016-04-09:
@rubert34 It has the _exact_ same behavior when the `nomodeset` option is enabled?

rubert34, 2016-04-10:
Yes, the result is the same now.
I made another video, where I try booting with normal, then with nomodeset.
https://youtu.be/SSxqWazq-dg

SevenBits, 2016-04-10:
@rubert34 Hmm, it's probably a hardware or driver issue then. Maybe contact the Kali Linux developers and see if they can help you?

SevenBits, 2016-05-14:
Closing due to no updates in over a month.

closed 2016-05-14 by SevenBits