SevenBits/Mac-Linux-USB-Loader #109: Persistence for Kali seems not to work

t-io, 2016-04-13:
Steps to reproduce: 
1. created a LiveUSB with Kali
2. setup persistence within the persistence-manager
3. Boot to Kali through advanced options with persistence enabled
4. saved some textfile on desktop
5. restart & reboot into kali through  advanced options with persistence enabled
6. file is gone. 

Does Mac-Linux-USB-Loader support persistence for Kali or is it ubuntu exclusive?

SevenBits, 2016-04-13:
As stated in the documentation, Mac Linux USB Loader only supports setting up persistence for Ubuntu, Linux Mint, Zorin, and elementary. Kali requires a more complicated procedure, namely, a partition on your USB drive.

_However_, one of my users has kindly written up a very nice tutorial showing how you can use persistence with Kali. It is not very complicated if you follow the directions. You can watch the [video][1] or view [the written instructions][2].

Again, I'm not affiliated with this individual nor do I endorse him, I'm just suggesting that you try his tutorial.

[1]: https://www.youtube.com/watch?v=JdV-0k7BHLk
[2]: https://pentestmac.wordpress.com/2015/11/28/kali-linux-usb-persistence-mac/

closed 2016-04-13 by SevenBits

t-io, 2016-04-14:
Thanks for your fast answer :) 
I must have overseen it in the documentation. The instructions work for me. 
Here is the correct [video link](https://www.youtube.com/watch?v=mDRbTHCoj8U) to the written instructions you suggested.

SevenBits, 2016-04-15:
@t-io Great, glad I could help! :+1: