SevenBits/Mac-Linux-USB-Loader #112: Confusing example of enterprise.cfg at https://sevenbits.github.io/Enterprise/

karlic, 2016-05-13:
**Short Description of Problem:** The example enterprise.cfg file at https://sevenbits.github.io/Enterprise/ does not appear to work with the current version of Mac Linux USB Loader. The one at https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Configure-booting-(one)-of-multiple-ISOs-(multiboot) **does** work.

**Expected Behavior:** 

Ability to boot from a kali.iso and/or a tails.iso file.

**Actual Behavior:** 

Booting fails.

(It seems that findiso=/efi/boot/boot.iso is wrong for the Tails and Kali entries and the options

    nopersistent noprompt timezone=Etc/UTC block.events_dfl_poll_msecs=1000 nox11autologin module=Tails noapic noapm nodma nomce nolapic nomodeset nosmp vga=normal-

are wrong for Tails.)

**Steps to Reproduce:** 

Copy kali.iso and tails.iso to /efi/boot/ on USB stick.

Copy enterprise.cfg from https://sevenbits.github.io/Enterprise/ to /efi/boot/enterprise.cfg

Try and boot from Tails and/or Kali.

SevenBits, 2016-05-14:
@karlic Thank you for your contribution, and for catching the error on the Enterprise website regarding the filename. I will correct this.

Regarding the kernel boot options: they were correct when I originally wrote the page with the help of Tails developers. They must have changed this, which is why the options aren't correct. I'd be happy to correct this, if you can verify that the fix indeed works with the latest version (that is official documentation, after all).

Also, thank you for the wiki article. I think I will make some small changes, and then link to it from the index page.

closed 2016-05-14 by SevenBits

karlic, 2016-05-14:
@SevenBits I can confirm that the options I have posted in https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/Configure-booting-(one)-of-multiple-ISOs-(multiboot) **do** work for tails-i386-2.3.iso and on a MacBook Pro (13-inch, Early 2011) running OSX 10.11.3.

They came from the source code for MLUL, so I can't really claim credit for it. ;-)

Lines 108--109 in SBEnterpriseConfigurationWriter.m

		} else if (family == SBDistributionTails) {
			[string appendString:@"kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live config live-media=removable noprompt timezone=Etc/UTC block.events_dfl_poll_msecs=1000 splash nox11autologin module=Tails quiet splash\n"];

SevenBits, 2016-05-14:
@karlic Alright, I'll double check the source and make the change. Thanks!