SevenBits/Mac-Linux-USB-Loader #113: Kali Download Broken (I Have a Fix)

learntofly83, 2016-05-17:
Kali Wont Download from v 3.2.1 due to the source being http rather than https

A hack for the end user is:

(I'm using sudo as I've changed the default permissions to root:admin on the .app)

`sudo nano /Applications/Mac\ Linux\ USB\ Loader.app/Contents/Info.plist `

Change the last few lines from: 

```
        <string>https://www.sevenbits.tk/appcasts/mlul.xml</string>
</dict>
</plist>
```

TO:

```
        <string>https://www.sevenbits.tk/appcasts/mlul.xml</string>
        <key>NSAppTransportSecurity</key>
                <dict>
                        <key>NSAllowsArbitraryLoads</key>
                        <true/>
                </dict>
</dict>
</plist>
```

Then run (assuming you have Xcode installed)

`sudo codesign -fs - /Applications/Mac\ Linux\ USB\ Loader.app`

I'm assuming the author can programatically build this into the next version

LTF

SevenBits, 2016-05-17:
@learntofly83 While this is a worthy hack, I'd like to avoid putting this into Mac Linux USB Loader. If, by the time of the next release, Kali _still_ has not enabled HTTPS for downloads, I will consider using a temporary exception to allow the app to use HTTP only with `kali.org`; other connections will still be forced to HTTPS.

learntofly83, 2016-05-17:
Sure it was more of a letting you know of the issue with Kali and an interim workaround for end users ;)

Unbelievable Kali isn't delivered via https isn't it!

SevenBits, 2016-05-18:
Yes, it's ironic that a distribution focused on security does not allow secure encrypted connections to their downloadable files.

commit 1cee4803 (2016-05-21): Add a temporary exception to allow downloading Kali over HTTP. Fixes #113.
closed 2016-05-21 by SevenBits