SevenBits/Mac-Linux-USB-Loader #114: custom kernel option (Number 9) doesn't seem to work.

dustingit, 2016-05-17:
**Short Description of Problem:** Under boot option 2 "Modify Linux kernel boot options" option "9" doesn't seem to work like other options.

**Expected Behavior:** 

If I choose option 9 it should change to yellow and then I should be able to hit "0" and hopefully specify an option like the initrd path so I can get different ISO's to work like ubuntu server

**Actual Behavior:** 
Doesn't change yellow and just hangs. Reboot required.

Boot to USB > Select option "2)" and then "9"

What I am trying to do is specify a custom initrd path so that I can try getting other ISO's to boot, like Ubuntu Server, or Pogostick. It would make this loader even more awesome. I have tried to specify the path in the .cfg file, which seemed to work, but get over written by some default configuration I am guessing.

Thanks for all your work thus far.

dustingit

SevenBits, 2016-05-18:
Works for me. Maybe it is working on your system and the prompt just isn't showing up for whatever reason?

dustingit, 2016-05-18:
Hey SevenBits,
Thanks for the response. I thought that maybe it working and just not
changing the text to yellow, but when I choose 9 and then 0 to boot, it
just hangs. I've tried it on two different Macs with the same results.

SevenBits, 2016-05-18:
You don't press 9 and then 0 -- you press 9 and type the additional options that you want to pass to the kernel. When you press Enter, those options will be saved and the text will turn yellow. Then you press 0 to boot.

dustingit, 2016-05-27:
Once I push the 9 it just sits there and doesn't respond to anything. I
will try the latest version and report back.

Thanks!

SevenBits, 2016-07-06:
I've received several reports on this. Will do some testing and report back.

SevenBits, 2016-08-14:
The issue should be resolved in the latest 3.3 beta; please try that. Closing for now.

closed 2016-08-14 by SevenBits

dustingit, 2016-08-18:
I am now  able to use option 9 and pass in kernel options. Thanks for working on it SevenBits!

SevenBits, 2016-08-19:
@dustingit No problem!