SevenBits/Mac-Linux-USB-Loader #115: MacBook Pro Retina support

ManuBermu, 2016-05-28:
**Short Description of Problem:** MacBook Pro Retina support

**Expected Behavior:** 

Unable to start Linux on a MacBook Pro Retina. An incompatibility message appears.

SevenBits, 2016-05-28:
Can you give more details? Or post a screenshot of this "incompatibility message".

ManuBermu, 2016-05-29:
Hi.
I tried to install Linux on a usb, complete installation with more than 4GB of persistence. And the mistake I obtained is as follows (image):
[image]

SevenBits, 2016-05-30:
Have you tried booting with the `nomodeset` option enabled (you can find it in the Advanced Settings) to rule out any graphical issues?

brettalton, 2016-08-02:
Following. I also have a 15" Macbook Pro Retina circa early 2013. I was able to get gParted installed via `vga=normal` so far, going to try Ubuntu right now.

SevenBits, 2016-12-19:
Closing due to lack of activity.

closed 2016-12-19 by SevenBits