SevenBits/Mac-Linux-USB-Loader #116: Keyboard and touchpad is not working on MacBook

bu7maid, 2016-05-28:
**Short Description of Problem:** When I load up Kali Linux using a usb, it works and loads up but then the keyboard and touchpad (mouse) of my MacBook (12-inch early 2015) does not work.

**Expected Behavior:** 

Keyboard and touchpad should work.

**Actual Behavior:** 

Keyboard and touchpad is not functioning

SevenBits, 2016-06-12:
Hmm, I’m not sure what could be causing this. Perhaps it’s a driver issue? Have a tried using an external mouse, such as a USB mouse or similar?

bu7maid, 2016-06-13:
I tried to use an external mouse as well as keyboard and it worked fine.

SevenBits, 2016-06-13:
@bu7maid Then an external mouse and keyboard worked, but not those on your Mac, it's a probably a driver issue. I'd file a bug report with Kali Linux.

jam4375, 2016-07-06:
I'm also seeing similar behavior with the 2016 12" MacBook. Both the keyboard and touchpad are not responding. This is while trying to install Xubuntu 16.04. I have posted a question here: 

http://askubuntu.com/q/794941/198090

Just to get it to boot I have to set `noacpi`. This [thread](http://askubuntu.com/q/794941/198090) suggests trying the option `nolacpi` instead. However, trying to choose 'Custom boot options' in the boot menu causes it to freeze.

The MacBook only has 1 USB-C port so I need to get a hub first to try an external keyboard and mouse while using a live USB.

Will hopefully report back,
Jason

SevenBits, 2016-07-06:
@jam4375 This has been observed by multiple different Linux distributions and it is an issue with your MacBook's hardware. Since your Mac is from this year it is probably too new and the folks who work on the Linux kernel have not gone about fixing the issue yet.

If you can use an external mouse and keyboard, I would do that for the time being.

Also, I am investigating issues running some parts of Enterprise on newer Macs. In the meantime, please enable the option by opening Mac Linux USB Loader, selecting the USB setup window, and clicking on the option to edit the Enterprise configuration file. You can then add that option manually.

closed 2016-08-14 by SevenBits

skylarmb, 2016-11-13:
Same problem on 2015 12 inch macbook. I've tried a number of other distros: Mint 18, Ubuntu 16.10, and Ubuntu 14.04.4 LTS

Cerise1, 2017-01-01:
Same problem with The new 15" macbook pro with touch bar ! It wont boot without nomodeset and acpi=off parameter and then when it boots keyboard and trackpad wont work.. If I add external usb keyboard or mouse it works... Any News ON this problem ?

gioferro96, 2017-05-31:
same problem on new macbook pro 13" late 2016 without touchbar!!

Hmachalani, 2017-08-18:
Still doesn't work for me. Macbook Pro 2016 with touchbar

mjsracing, 2017-08-23:
Same issue. MacBook Pro 2016 with touchbar

LeonBuchner, 2017-09-05:
Same here!
I have tried to create a Kali Linux Boot Stick -> Boot was without problems -> After Boot Keyboard and Touchpad don't work.

MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports), 2.9 GHz Intel Core i5, 8 GB 2133 MHz LPDDR3, Intel Iris Graphics 550 1536 MB

Afrika90, 2017-10-10:
Same problem with MacBook 12 early 2015. Trackpad and mouse doesn’t work. I’ve read About a driver that it’s going in conflict with the apple hardware. Any news about that?

demuze28, 2017-10-16:
Still having the problem with the new macbook and parrot OS, anyone know when they will get around to fixing this?

yougz11, 2017-10-30:
Hi all,
I have the same troubles with MBP 13" without touchbar 2017 (2 x USB-C ports)
Does somebody have found a solution ?

bbgun7, 2017-11-05:
I'm having the same problem with my Macbookpro12,1 after upgrading to os x high sierra on the os x side of my dualboot. The keyboard works in refind, and the grub menu of live usbs, but otherwise only external keyboards work

Hodges3902, 2017-11-07:
Same problem here... MacBook pro 2017

yougz11, 2017-11-07:
If you guys need a solution : try to install on virtual machine. I m under vmware fusion 10 and kali linux and its apps works perfectly with native keyboard and trackpad of mbp 2017.... this is the only solution i found. I ve posted a bug report on kali linux ‘s website and they answered me sayin’ that there is no chance for mbp pro owners that the solution will be included in the next version. If some of you want to see their complete answer, let me know....

jessingrass, 2017-11-28:
Same problem here with 2015 macbook air and Ubuntu 17.10. I'm going to see if an issue has been filed with ubuntu.

Shatterblade, 2018-01-04:
You might have to use a custom kernel that includes Mac Drivers, or find a distro that includes them. (Don't ask me how to do it; I have no idea)

silasabbott, 2018-01-14:
Same problem here on my **MacBook Pro 2017, 15-inch**. Also gotta figure out how to get the escape button to show up on the TouchBar, no way to move back up to the previous menu!

dastoiboy, 2018-01-15:
I am having the same problem with a 2017 Mac Book Pro. When I connect an external mouse it works and a keyboard, but I can only plug in either or. And the entire point is for it to be portable. So does anyone have a fix for this?

silasabbott, 2018-01-16:
### **[macbook12-spi-driver](https://github.com/cb22/macbook12-spi-driver)**
[State of Linux on the MacBook Pro 2016 & 2017](https://github.com/Dunedan/mbp-2016-linux) by Dunedan might have what we're looking for. Under the _[Keyboard & Touchpad](https://github.com/Dunedan/mbp-2016-linux#keyboard--touchpad)_ section, a driver is mentioned:

> Basic keyboard and touchpad functionality is working with the out-of-tree driver developed for the Retina Macbook (https://github.com/cb22/macbook12-spi-driver).

Anyone have a chance to try this **[macbook12-spi-driver](https://github.com/cb22/macbook12-spi-driver)** on Kali?

Basic TouchBar support is also mentioned [here](https://github.com/Dunedan/mbp-2016-linux#touch-bar).

rcorre, 2018-01-16:
@silasabbott It worked for me with a 13,2 macbook pro running archlinux. Unfortunately, the "not working" state of the wifi card is absolutely true, so the machine is still mostly useless ...

cRaZyPaNd4, 2018-03-03:
same issue
macbookpro 2016 running kali linux
trackpad and keyboard not functioning

djsan15, 2018-04-06:
Same issue. Still not working on 13" MBP with Touchbar

afrika1990, 2018-04-06:
Guys, i've solve the problem, u must re-compile ur kernel, and remove one driver when i'll do it. Also u can found asolution here on github with how u could re-compile the kernelin a correct mode.

owaisafaq, 2018-05-03:
@afrika1990 : Can you provide details? I've been waiting like 1000s of other users for a solution to this problem.

afrika1990, 2018-05-03:
https://github.com/cb22/Macbook 12-spi-driver

Here the link

george12341, 2019-03-31:
Did anybody find a solution about this issue? I created bootable Kali Linux using a USB on my macOS mojave but when I trynna select a language, my mac's keyboard and touchpad do not work.

rcorre, 2019-03-31:
The link mentioned in @silasabbott's [comment](https://github.com/SevenBits/Mac-Linux-USB-Loader/issues/116#issuecomment-357846170) worked for me (but again, a working keyboard only got me far enough to hit wifi driver issues).

zloykoteyka145, 2019-04-01:
Same issue. Macbook 12" 2017 Late

JayOceans, 2019-05-06:
same problem here. macbook 12" early 2015

yeah-right, 2019-05-18:
Same problem. macbook pro 13" early 2019 :\

VGamezz19, 2020-02-29:
same problem here macboock 12 " 2016.

kgbook, 2020-09-26:
@MiladMass Dual boot is ok. [refind](https://sourceforge.net/projects/refind/) works for me.

kgbook, 2020-09-26:
Built-in touchpad is not working on my MBP17, but keyboard is ok.
Did anybody find a solution about this issue?

sasdallas, 2020-09-27:
I have a MacBook Air 2012, keyboard is fine but touchpad is not working.

Anyone find a fix?

jylescoad-ward, 2021-09-20:
Currently having the exact same issue on my MacBookPro12,1

What's weird is that this is only happening on 3rd party operating systems (any linux distro :/)

Any solutions yet or still radio silence?

senyaa1, 2021-09-25:
@jylescoad-ward My keyboard and touchpad now works on Arch Linux. But audio is working only through headphones, speakers don't work.

jylescoad-ward, 2021-09-25:
> @jylescoad-ward My keyboard and touchpad now works on Arch Linux. But audio is working only through headphones, speakers don't work.

Lucky! I tried to get arch Linux working but even the liveusb didn't detect my keyboard and touchpad

aayushambekar20, 2022-02-04:
MacBook Air 2020 same problem please someone give a solution

mtmak9, 2022-03-14:
Hey Guys, I had same problem like you with Kali Linux 2019 ver. On my Mac is doesnt work keyboard and Mouse. But I check the latest Kali distro from 2022, and It works.
They fixed this problem. I am Macbook Pro 2016 (Sulfur) 

So if you have problem with working keyboard&mouse, just check new Kali from 2022. Should works.

Xentuous, 2022-03-23:
hi @mtmak9, did u install it into a USB which u then dual boot it from your mac? If so, did u download the "kali-linux-2022.1-live-everything-amd64.iso"?

BJW333, 2022-07-11:
same fucking issue idk what the deal is I live usb booted to my 15" Mac i7 monetary its annoying as hell