SevenBits/Mac-Linux-USB-Loader #117: Kali 2.0: no controller found Mac Pro 13 Retina

JustSeb, 2016-06-06:
Tried to install Kali Linux Live from Usb, when i choose to boot from the USB this is what i get on the screen:

[    1.899668] i8042: No controller found
Loading, please wait...
[    2.175150] usb usb2-port1: couldn't allocate usb_device
[   12.589508] usb 1-5: device descriptor read/all, error -110

BusyBox v1.22.1 (Debian 1:1.22.0-9+deb8u1) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/bin/sh: can't access tty: job control turned off
(initransfs) Unable to find a medium containing a live file system
modprobe: module ehci-orion not found in modules.dep

SevenBits, 2016-06-07:
Please try booting with `nomodeset`.

JustSeb, 2016-06-07:
tried already, same thing happens .. also tried to turn of encryption, found a forum where it helped to someone..but nothing..

SevenBits, 2016-06-07:
Maybe it's a driver issue? Can you try on another Mac and see if it works there?

ManuBermu, 2016-06-12:
I have the same problem. I try on Mac Mini 2011 and it's work but on MacBook Pro Retina no.

SevenBits, 2016-06-12:
@ManuBermu, can you try another distribution, such as Ubuntu? I’d like to know if this is a Kali-specific issue; if it is, then you need to take this up with them.

ManuBermu, 2016-06-13:
I try with Elementary OS, Ubuntu, Kali and the only distribution that works 100 % is Tails using its own installer.

SevenBits, 2016-06-13:
@ManuBermu, that's interesting. I wonder what Tails is doing that Mac Linux USB Loader isn't... I will try to setup Tails myself, perhaps something has changed in recent versions.

JustSeb, 2016-06-13:
yep.. Tails works for me two..

SevenBits, 2016-06-14:
@JustSeb, what did you do to get Tails to work?

JustSeb, 2016-06-14:
create live usb, persistence manager and reboot.. just the standard procedure

glasshapful, 2016-06-15:
I was able to successfully boot from a flash drive using the previous version (v3.2.1) of Mac-Linux-USB-Loader, but the latest version (v3.2.2) gave me the "no controller found" error for both Debian and Ubuntu on a late-2012 mac mini and an early-2015 macbook pro.

SevenBits, 2016-06-15:
@glasshapful That's strange, nothing related to the configuration was changed in those versions.

JustSeb, 2016-06-15:
tried the 3.2.1 version, same result.. :/

SevenBits, 2016-12-19:
Closing due to lack of activity.

closed 2016-12-19 by SevenBits