SevenBits/Mac-Linux-USB-Loader #118: Linux Mint Installation failure: ISO lacks an EFI-enabled kernel

g5mini, 2016-06-09:
Hello, please bare with me, I'm new to Linux. I'm an end user on the Mac platform, and a GUI user at that. But, I found the Mac Linux USB installer, and decided to install it, in order to format a USB flash drive - to install linuxmint-17.3-cinnamon-64bit.

I downloaded linuxmint-17.3, erased the USB stick using Disk Utility in Mac OS 10.11.5, and ran the boot installer, locating the Linux Mint ISO file. However, when the ML-USB program gave me the options from the drop down menu, there was no Linux Mint as an option for the distribution family. "Ubuntu" was the first one in the list, and the window looked like this: 
[image]

So, I chose "other". I began the installation, and after it completed, I tried to reboot from the USB flash drive, and saw this at startup in Mac OS: https://ibin.co/2juJlZR7T7Ji.jpg
When I selected EFI boot, I came into a DOS like screen, and had the following presented to me: 
1: https://ibin.co/2juLO6Ycn44g.jpg 
2: https://ibin.co/2juK2Qk5n2Vu.jpg  
I tried several times selecting option 1, but whenever I did, I got this: https://ibin.co/2juLzZSZ3jiM.jpg 

At some point, I think around the 3rd attempt, I got this screen, although I'm not sure how I got to it: 
https://ibin.co/2juLpGf55ME6.jpg, but when I selected this option, I think I got the same: 
No Medium Found: https://ibin.co/2juLzZSZ3jiM.jpg 

I'm definitely missing something. Maybe I download the wrong ISO. From the beginning, based on this tutorial: http://www.makeuseof.com/tag/how-to-boot-a-linux-live-usb-stick-on-your-mac/ — I never once saw "Linux Mint" in the drop down menu for available distro's.  

I'm at a complete standstill with this, I really want to check out Linux Mint. 
Please help. Thanks.

SevenBits, 2016-06-09:
@g5mini Hi there. Linux Mint is a distribution that is based on Ubuntu, so Mac Linux USB Loader uses Ubuntu's settings to boot it. Mac Linux USB Loader should select the distribution family for you based on the ISO file that you give it. In this case, it (correctly) chose Ubuntu because it saw that you were using Linux Mint. Please re-do the procedure and leave Ubuntu selected as the distribution family.

P.S.: Kudos for your well-formatted post.

g5mini, 2016-06-09:
Thank you for such a quick response. Can I ask...
When I do select Ubuntu as the distro from the menu (although it automatically appears at the top of the list), when I reboot into the Enterprise terminal window, should I select: 

Option 1 ?: Welcome To Enterprise - Version 0.3.2 - Available Boot Options: 
https://ibin.co/2juLO6Ycn44g.jpg — Boot Linux from ISO file? 

Next, if I get to this screen again: Start Linux Mint: https://ibin.co/2juLpGf55ME6.jpg 
Should I choose the first option in the list: Start Linux Mint 17.3 Cinnamon 64-bit? 

Here's a question to throw into the mix, I was under the impression that when I first formatted the USB flash drive, that I could choose exFAT as the partition scheme - only because from what I've read, exFAT supports more hardware, and is basically the newer version of FAT32. 

As an end goal, I'd eventually like to install a Linux Distro on an slightly larger external HD (120Gb's), that will run natively on my Mac's hardware without any firmware updates required; a distro that has more media software available for that particular distro. 

As a side note, I'm only really interested in Linux, to try and learn how it's constructed, how it works. The Mac OS platform is very similar in some ways, but I want to be able to poke around, learn, play, make mistakes in a Linux distro on a separate HD, so I don't muck up my Mac OS' Kernel if I screw something up in Terminal.

Again, thanks so much for you reply, I'll follow your instructions.

SevenBits, 2016-06-09:
Hi @g5mini, the first thing you should do is read [the official instructions on the wiki](https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/General-Instructions), if you have not already. These instructions are more up-to-date than some of the others you might find online.

You can select option 1 or option 2 from the main menu; it doesn't matter. Option 2, however, lets you enable some debug/compatibility options in case Linux does not work on your computer.

That second menu, with the whole "Start Linux Mint" thing, is the fallback menu provided by the distribution. That shouldn't show up unless there's a problem finding the kernel. Rather than focus on that, I'd instead make sure you're selecting the right options in Mac Linux USB Loader.

exFAT vs FAT32 doesn't matter, but some Linux distributions don't like exFAT. Your Mac should be able to boot from your USB stick either way, so it depends on what works for you.

g5mini, 2016-06-09:
Thanks again, I think part of what is confusing me about this process is the first image in my first post.
"This ISO lacks an EFI-enabled kernel" That check box was automatically selected. 

I guess the frustration comes from not knowing what that means - if you're new to this process as I am. The dialog box doesn't explain what "lacks an EFI-enabled kernel" means, or what it means to have that option un-selected - with a description for each.

g5mini, 2016-06-09:
Got Linux Mint to boot into the desktop. What an odd OS. It looks and behaves kind of like... Windows NT, crossed with... not sure. Not super refined, but very usable I guess. I'm so used to the Mac GUI. 

I'll definitely poke around in there at some point. Thanks for your help, it worked.

SevenBits, 2016-06-09:
If you'd prefer a more Mac-like experience, you could use Elementary OS. You can download it from the distribution downloader. And there's always standard Ubuntu...

closed 2016-06-09 by SevenBits