SevenBits/Mac-Linux-USB-Loader #120: No wifi and do I need to do a full install after I boot into live CD?

chovy, 2016-06-30:
**Short Description of Problem:** put text here

No wifi (MacOS) and do I need to do a full install after I boot into live CD?

**Expected Behavior:** 

Expect wifi to work when I boot into live disk from USB.

**Actual Behavior:** 

No wifi

**Steps to Reproduce:** 

Follow video here: https://www.youtube.com/watch?v=tEC89FkXPE8

Also I'm wondering if to do a persistent install do I need to start the Install Ubuntu process by clicking the icon on the desktop? I tried this and it asks to install to /dev/sdb2 but it never finishes. Don't know if this is because I have no wifi or what.

SevenBits, 2016-07-04:
Whether or not WiFi works depends entirely on your distribution and which Mac you have. Some Macs require proprietary drivers that do not ship by default with most Linux distributions. You may need to install them manually.

You don't need to actually install Ubuntu to have a persistent install. You just run the persistence manager tool in Mac Linux USB Loader and select the appropriate option in the boot manager.

chovy, 2016-07-04:
@SevenBits hey, so I got it working w/ a live usb drive. But I want to install from the live usb to another usb as a regular full install. I went through that process (using two usb drives) but after installation was completed, I could not boot into the new install from my usb drive.

I think the xubuntu installer wiped out the partitions that your software created. Thus making it unbootable. I read somewhere to use advanced partitioning, but I'm not sure what to do here. Should I use Fat-32? I am trying with Ext4 now...but I'm guessing this isn't going to work for some reason.

I don't see any `./efi` directory when I mount the blank drive.

SevenBits, 2016-07-05:
@chovy Mac Linux USB Loader is designed to allow you to create bootable live USBs. Once you boot into Ubuntu, it's all you in terms of the installation. I can't provide Ubuntu installation tips; there are a number of other good resources online for that.

To get you started, though, perhaps you should not try to install over the files added by Mac Linux USB Loader. It's also possible that the XUbuntu installer installs Ubuntu in a way that the Mac doesn't recognize.

Hope this helps.

closed 2016-07-05 by SevenBits