SevenBits/Mac-Linux-USB-Loader #121: Program doesn`t recognize GUID format USB thumbdrives

garotadpi, 2016-07-07:
At the choose Destination box, USB thumbdrives formatted with the Mac OS file system with the GUID partition table don`t appear on the Choose Destination Box.

**Expected Behavior:** 

A thumbdrive formatted with the Mac OS file system anda GUID partition table is inserted in the computer, pressing "Refresh" on the Choose Destination box should make it appear on the box, as a selectable USB Drive to install the Live Linux distribution.

**Actual Behavior:** 

A thumbdrive formatted with the Mac OS file system anda GUID partition table is inserted in the computer, pressing "Refresh" on the Choose Destination box has no effect. Only FAT32 formatted thumbdrives appear.

**Steps to Reproduce:** 

Format a thumbdrive with the Disk Utility App to use the Mac OS file system. Open the Mac Linux USB Loader. Choose Linux Distribution to be installed on the thumbdrive. Press "Refresh" to make the Mac OS formatted thumbdrive appear on the Choose Destination box.

SevenBits, 2016-07-07:
Have you enabled the "allow HFS+ formatted drives" option in the preferences?

Furthermore, the documentation states that a FAT-formatted drive is required, so this isn't exactly a bug.

SevenBits, 2016-07-20:
Closing due to no activity.

closed 2016-07-20 by SevenBits