SevenBits/Mac-Linux-USB-Loader #122: Immediate crash upon trying to download a distro

drfraser, 2016-07-15:
Hi,

I have OS X 10.8.5, iMac 2010....  I am using Mac Linux USB Loader v3.2.2.  Nothing special about my system as far as i know

As soon as I click on any distro in the Distribution Downloader screen e.g. Ubuntu 16.04, the app crashes.  Where does it put the downloaded ISOs?

Process:         Mac Linux USB Loader [991]
Path:            /Users/USER/*/Mac Linux USB Loader.app/Contents/MacOS/Mac Linux USB Loader
Identifier:      com.sevenbits.Mac-Linux-USB-Loader
Version:         3.2.2 (9405)
Code Type:       X86-64 (Native)
Parent Process:  launchd [371]
User ID:         501

Date/Time:       2016-07-15 18:03:07.734 +0100
OS Version:      Mac OS X 10.8.5 (12F2560)
Report Version:  10

Interval Since Last Report:          333443 sec
Crashes Since Last Report:           11
Per-App Interval Since Last Report:  258 sec
Per-App Crashes Since Last Report:   4
Anonymous UUID:                      78C158A8-EA96-3A2E-724E-A5A954604ACD

Crashed Thread:  6  Dispatch queue: com.apple.root.default-priority

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[__NSCFConstantString stringByAddingPercentEncodingWithAllowedCharacters:]: unrecognized selector sent to instance 0x109d4f8e0'
terminate called throwing an exception
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff924f3b06 __exceptionPreprocess + 198
1   libobjc.A.dylib                     0x00007fff909623f0 objc_exception_throw + 43
2   CoreFoundation                      0x00007fff9258a40a -[NSObject(NSObject) doesNotRecognizeSelector:] + 186
3   CoreFoundation                      0x00007fff924e202e ___forwarding___ + 414
4   CoreFoundation                      0x00007fff924e1e18 _CF_forwarding_prep_0 + 232
5   Mac Linux USB Loader                0x0000000109d0d8cf Mac Linux USB Loader + 18639
6   libdispatch.dylib                   0x00007fff95a6ef01 _dispatch_call_block_and_release + 15
7   libdispatch.dylib                   0x00007fff95a6b0b6 _dispatch_client_callout + 8
8   libdispatch.dylib                   0x00007fff95a6c1fa _dispatch_worker_thread2 + 304
9   libsystem_c.dylib                   0x00007fff8f3ebcdb _pthread_wqthread + 404
10  libsystem_c.dylib                   0x00007fff8f3d6191 start_wqthread + 13

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff92bd7686 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff92bd6c42 mach_msg + 70
2   com.apple.CoreFoundation      	0x00007fff92490233 __CFRunLoopServiceMachPort + 195
3   com.apple.CoreFoundation      	0x00007fff92495916 __CFRunLoopRun + 1078
4   com.apple.CoreFoundation      	0x00007fff924950e2 CFRunLoopRunSpecific + 290
5   com.apple.HIToolbox           	0x00007fff96e28eb4 RunCurrentEventLoopInMode + 209
6   com.apple.HIToolbox           	0x00007fff96e28c52 ReceiveNextEventCommon + 356
7   com.apple.HIToolbox           	0x00007fff96e28ae3 BlockUntilNextEventMatchingListInMode + 62
8   com.apple.AppKit              	0x00007fff8e692533 _DPSNextEvent + 685
9   com.apple.AppKit              	0x00007fff8e691df2 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 128
10  com.apple.AppKit              	0x00007fff8e6891a3 -[NSApplication run] + 517
11  com.apple.AppKit              	0x00007fff8e62dbd6 NSApplicationMain + 869
12  libdyld.dylib                 	0x00007fff900a07e1 start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff92bd9d16 kevent + 10
1   libdispatch.dylib             	0x00007fff95a6ddea _dispatch_mgr_invoke + 883
2   libdispatch.dylib             	0x00007fff95a6d9ee _dispatch_mgr_thread + 54

Thread 2:: com.apple.NSURLConnectionLoader
0   libsystem_kernel.dylib        	0x00007fff92bd7686 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff92bd6c42 mach_msg + 70
[... 335 lines cut ...]
__TEXT                              152.5M
__UNICODE                             544K
mapped file                          45.3M
shared memory                         512K
===========                        =======
TOTAL                                 1.4G
TOTAL, minus reserved VM space      444.9M

Model: iMac11,2, BootROM IM112.0057.B01, 2 processors, Intel Core i3, 3.2 GHz, 4 GB, SMC 1.64f5
Graphics: ATI Radeon HD 5670, ATI Radeon HD 5670, PCIe, 512 MB
Memory Module: BANK 0/DIMM1, 2 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54313235533654465238432D48392020
Memory Module: BANK 1/DIMM1, 2 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54313235533654465238432D48392020
AirPort: spairport_wireless_card_type_airport_extreme (0x168C, 0x8F), Atheros 9280: 4.0.72.0-P2P
Bluetooth: Version 6.1.7f5 15859, 3 service, 21 devices, 3 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: WDC WD1001FALS-40Y6A0, 1 TB
Serial ATA Device: HL-DT-STDVDRW  GA32N
USB Device: hub_device, 0x0424  (SMSC), 0x2514, 0xfa100000 / 2
USB Device: DataTraveler 2.0, 0x0951  (Kingston Technology Company), 0x1665, 0xfa130000 / 5
USB Device: BRCM2046 Hub, 0x0a5c  (Broadcom Corp.), 0x4500, 0xfa110000 / 4
USB Device: Bluetooth USB Host Controller, apple_vendor_id, 0x8215, 0xfa111000 / 6
USB Device: Internal Memory Card Reader, apple_vendor_id, 0x8403, 0xfa120000 / 3
USB Device: hub_device, 0x0424  (SMSC), 0x2514, 0xfd100000 / 2
USB Device: Built-in iSight, apple_vendor_id, 0x8502, 0xfd110000 / 4
USB Device: IR Receiver, apple_vendor_id, 0x8242, 0xfd120000 / 3

SevenBits, 2016-07-15:
Could you try again with the latest 3.3 beta version to see if the problem persists there?

drfraser, 2016-07-15:
I did the exact same steps and it still crashes with the same exception. I'd take a crack at it, to see if I can find the line, but have to update XCode and all that first.

SevenBits, 2016-07-16:
@drfraser Hello there -- does the error occur when you do a _single click_ or the distribution name, or when you do a _double click_?

closed 2016-07-16 by SevenBits

SevenBits, 2016-07-20:
@drfraser This has been fixed.

drfraser, 2016-07-22:
thanks. But I can't compile until I figure out how to workaround IB_DESIGNABLE as XCode 5 does not support that etc - and the compiled versions (3.3, 3.2) I can download haven't been updated (?).  If no one else is trying to run this on 10.8, then this is entirely my problem and I'll figure it out :)

SevenBits, 2016-07-22:
@drfraser I’m doing everything I can to get the update out as soon as possible. If you have a pressing need for a compiled build with the fix you can [contact me](https://www.sevenbits.tk/contact.html) and we can work something out.

drfraser, 2016-07-23:
My apologies for seeming to be impatient; I didn't intend that. I shouldn't have left a note; my compiling problems are not relevant to the issue.

SevenBits, 2016-08-15:
@drfraser Updated version with fix is out.

commit cb75c876 (2016-07-16): Fix the distribution downloader on 10.8 (fixes #122).

The Wikipedia search feature was calling a function that was 10.9+ only, and this was causing a crash on 10.8. The code has been patched to use an older function on 10.8.