SevenBits/Mac-Linux-USB-Loader #123: Knoppix (Debian-based), doesn't install or boot properly

brettalton, 2016-08-02:
**Short Description of Problem:** Using KNOPPIX_V7.6.1DVD-2016-01-16-EN.iso, I tried installing Knoppix as a liveUSB and it doesn't appear to work. Files aren't copied and USB won't boot.

**Expected Behavior:** 

Install Knoppix, reboot, boot into Knoppix LiveUSB.

**Actual Behavior:** 

When installing Knoppix as LiveUSB, 'enterperise.cfg' pops up asking me to manually configure my bootloader. I can't manually copy the ISO over either because the drive is formatted as FAT.

**Steps to Reproduce:** 

Wipe USB stick as FAT or exFAT (tried both) using Disk Utility. Download latest Knoppix: http://ftp.uni-kl.de/pub/linux/knoppix-dvd/. Try installing via Mac Linux USB Loader. See enterprise.cfg pop up after installation. Note that no files appear to be copied to the USB stick. Reboot, hold option/alt, try booting, notice failure.

[image]

brettalton, 2016-08-02:
This menu also doesn't appear when trying to install Knoppix (note: this is an Ubuntu ISO). It just goes from choose destination / enterprise to install.
[image]

SevenBits, 2016-08-02:
@brettalton Knoppix is not supported by default in Mac Linux USB Loader, so that is why the Enterprise settings window comes up; you need to fill out the parameters.

In regards to your second screenshot, can you be more specific as to what you mean? What do you mean it ‘skips the menu’?

brettalton, 2016-08-04:
I don't mind filling out the parameters, but what I'm saying is that the program doesn't detect that FAT32 doesn't support 4GB+ files, so it gives the appearance that the installation was successful, but stick doesn't contain the knoppix.iso. The program thus fails to throw a warning and the user is unaware of the failure, reboots and notes that there is no OS to load.

As for the second screenshot, I was completely incorrect in stating that it skips the menu.
[image]
[image]
[image]
[image]

SevenBits, 2016-08-04:
@brettalton Ah, my mistake! Yes, I see what you mean now. I'll add an error message that gets displayed if the ISO is larger than 4 GB. Alternatively, you could try enabling the option to install on an HFS+ formatted drive; that would let you use larger file sizes, and is detected at boot by Macs (but, alas, only by Macs).

(Honestly, I can't believe it's that big. The largest ISO I've seen is Kali, at close to 2.5 GB; large, but still well below the 4 GB limit.)

brettalton, 2016-08-04:
Continuing your side note, I completely agree re: Knoppix size. I saw old screenshots of me performing data recovery with Knoppix 5.1.1 which was a 2007 distro at ~700MB and it worked just great. No longer having a CD-ROM drive of course leads me to using USB sticks. Using a 2014 Mac, I assumed I needed a recent distro and it appears Knoppix stopped with CD-sized ISOs in 2013 and now only have ~4GB ISOs available in 2015 and 2016.

http://ftp.uni-kl.de/pub/linux/knoppix/ (last CD ISOs)
http://ftp.uni-kl.de/pub/linux/knoppix/DVD/ (latest ISOs)

I ended up using Ubuntu 16.04 just fine. And if you're curious, it wasn't a important data recovery, just an old external hard drive that was ext3 and I wanted to transfer to a NTFS hard drive that I use on my Mac.

SevenBits, 2016-08-04:
@brettalton I'm glad you got it working. I'll add that warning you mentioned.

closed 2016-08-04 by SevenBits

zbeekman, 2017-08-07:
I would love to know what the correct parameters to fill in for the enterprise config are for Knoppix. Poking around I found this file named `isolinux.cfg`: 

```
# KNOPPIX 7.7 isolinux/syslinux boot config

DEFAULT auto
APPEND lang=en apm=power-off initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1

TIMEOUT 100
# KBDMAP german.kbd
PROMPT 1
F1 boot.msg
F2 f2
F3 f3
DISPLAY boot.msg

LABEL auto
KERNEL ifcpu64.c32
APPEND knoppix64 -- knoppix

LABEL knoppix
KERNEL linux
APPEND lang=en apm=power-off initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1

LABEL adriane
KERNEL linux
APPEND lang=en apm=power-off video=640x480 initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1 adriane

LABEL knoppix64
KERNEL linux64
APPEND lang=en apm=power-off initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1

LABEL adriane64
KERNEL linux64
APPEND lang=en apm=power-off video=640x480 initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1 adriane

LABEL debug
KERNEL linux
APPEND lang=en apm=power-off initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1 debug

LABEL debug64
KERNEL linux64
APPEND lang=en apm=power-off initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1 debug

LABEL grub
KERNEL grub.exe
APPEND --config=;

LABEL fb1024x768
KERNEL linux
APPEND lang=en apm=power-off vga=791 video=1024x768 xmodule=fbdev initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1

LABEL fb1280x1024
KERNEL linux
APPEND lang=en apm=power-off vga=794 video=1280x1024 xmodule=fbdev initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1

LABEL fb800x600
KERNEL linux
APPEND lang=en apm=power-off vga=788 video=800x600 xmodule=fbdev initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1

LABEL fb640x480
KERNEL linux
APPEND lang=en apm=power-off vga=785 video=640x480 xmodule=fbdev initrd=minirt.gz nomce libata.force=noncq hpsa.hpsa_allow_any=1 loglevel=1

LABEL memtest
KERNEL memtest
APPEND foo

LABEL dos
KERNEL memdisk
APPEND initrd=balder.img

LABEL failsafe
KERNEL linux
APPEND lang=en vga=normal atapicd nosound noapic nolapic noacpi pnpbios=off acpi=off nofstab noscsi nodma noapm nousb nopcmcia nofirewire noagp nomce libata.force=noncq hpsa.hpsa_allow_any=1 nonetwork nodhcp xmodule=vesa initrd=minirt.gz
```

Also, here is the structure of the Knoppix ISO:

```
⏚ [ibeekman:/Volumes/KNOPPIX 1] 2s $ tree
.
├── KNOPPIX
│   ├── KNOPPIX
│   ├── KNOPPIX1
│   ├── LICENSE.txt
│   ├── README_Security.txt
│   ├── SOURCES.txt
│   ├── background.jpg
│   ├── background_source_apophysis.flame
│   ├── bootonly.iso
│   ├── bootonly.iso.md5
│   ├── index_de.html
│   ├── index_en.html
│   ├── index_es.html
│   ├── index_fr.html
│   ├── index_it.html
│   ├── knoppix-cheatcodes.txt
│   ├── knoppix-logo-medium.png
│   ├── knoppix-logo-small.png
│   ├── kversion
│   ├── modules
│   ├── overlay-contents.txt
│   └── sha1sums
├── autorun.bat
├── autorun.inf
├── autorun.pif
├── boot
│   └── isolinux
│       ├── balder.img
│       ├── boot.cat
│       ├── boot.msg
│       ├── bootefi.msg
│       ├── f2
│       ├── f3
│       ├── german.kbd
│       ├── gfxboot.c32
│       ├── grub.exe
│       ├── ifcpu64.c32
│       ├── isolinux.bin
│       ├── isolinux.cfg
│       ├── ldlinux.c32
│       ├── libcom32.c32
│       ├── libmenu.c32
│       ├── libutil.c32
│       ├── linux
│       ├── linux64
│       ├── logo.16
│       ├── memdisk
│       ├── memtest
│       ├── menu.c32
│       ├── minirt.gz
│       ├── syslnx32.cfg
│       ├── syslnx64.cfg
│       ├── vesa.c32
│       ├── vesainfo.c32
│       └── vesamenu.c32
├── cdrom.ico
├── efi
│   └── boot
│       ├── BOOTIA32.efi
│       ├── BOOTX64.efi
│       ├── hashtool.efi
│       ├── keytool.efi
│       ├── ldlinux.e32
│       ├── ldlinux.e64
│       └── loader.efi
└── index.html

6 directories, 60 files
```

@SevenBits (and @brettalton and anyone else who knows) how do I specify the kernel and initrd in this case? This is what I have, but it looks dubious:

```
# enterprise.cfg
#
# This file is used to configure Enterprise. You need to fill out the following parameters according to how your desired Linux distribution configures its ISO file.

entry Knoppix
family Debian
iso boot.iso
kernel /boot/isolinux/linux64
initrd /boot/isolinux/minirt.gz

# Please see https://sevenbits.github.io/Enterprise/ for more information.
```

commit b36f4411 (2016-08-04): Warn the user if their file can't fit on a FAT32 USB stick (fixes #123).