SevenBits/Mac-Linux-USB-Loader #124: Cant see an EFI usb disk to boot from at the mac boot drive selector (option key boot).

cems2, 2016-08-03:
**Short Description of Problem:** put text here

was unable to get the mac to boot with any of 3 distros on 3 different USB sticks, trying all USB ports.

**Expected Behavior:** 

after creating the sticks on one mac (running e-capitan) I put them into the core2duo mad and held the option key down as I booted.  I expected the USB stick to appear in the menu selector for the boot drive.  

**Actual Behavior:** 

The boot menu  showed only the main HDD and the Recovery Disk (part of the 10.6 OS install). there was no other option visible.

**Steps to Reproduce:** 

created USB sticks for 3 different 64 bit distros but they would not appear at boot time on a macbook 2006 core 2 duo that has 10.6 OSX installed.

the sticks were created on a different computer (an i3 running el capitan).

I tried 3 different USB sticks
I tried 3 different distros, all 64 bit
puppy Linux tahr (ubuntu derived)
macpup (based on puppy linux)
lubuntu6.0.5 amd 64 (a light weight version of ubintu) 

I tried selecting the EFI check box on or off in the case of lubuntu with no change in outcome.

SevenBits, 2016-08-03:
The [system requirements][1] state that a requirement is:

> A Mac or PC with 64-bit UEFI firmware (any Mac made since 2007 should work)

You have a Mac with 32-bit UEFI firmware, which is not supported by default. You can build [the boot loader][2] from source though on 32-bit Linux, and that will create binaries that will boot on your Mac.

[1]: https://www.sevenbits.tk/mlul/sys-requir.html "System Requirements"
[2]: https://github.com/SevenBits/Enterprise "Enterprise"

closed 2016-08-03 by SevenBits

cems2, 2016-08-04:
Thank YOU!  Would something like Refid be another solution here?

And will the boot loader built that be 64 bit linux compatible.

Perhaps add to your suggestion heap of having your tool warn the user if the hardware is older than 2007.  That of course would not catch the cases built on another machine.

SevenBits, 2016-08-04:
rEFInd is more geared towards serving as a traditional boot manager for dual booting Linux on Macs (or other EFI-based systems, really), not for creating bootable USB sticks, so as far as I know the answer there is no.

Yes, the boot loader built _should_ be 64-bit Linux compatible, though it will not boot on systems with a 64-bit EFI. Almost every system on the planet that uses EFI uses 64-bit EFI with a 64-bit OS; both there are a few Mac models (such as yours) which use a 32-bit EFI with a 64-bit OS. Why Apple did this I don't know, but it complicates things somewhat. Be sure to build both components -- Enterprise (the boot manager) and GRUB. You can find both sets of sources at that link.

Can hardware older than 2007 even run Mac Linux USB Loader? I know the current minimum requirement is 10.8, and 10.8 will likely be dropped soon anyway, so I don't think it will be worth it.

SevenBits, 2016-08-04:
@cems2 Sorry, at some point I will write up much better instructions for how to build the boot manager.

The process is actually quite simple:

> 1. Download the source code tarball (or clone it from git). You can find the source code and binaries on the [releases page][1].
> 1. Extract the file and you'll get a source directory. Open a shell prompt (terminal) in this directory.
> 1. Run the script `install-deps.sh` to install the dependencies.
> 1. Run `build.sh` to automate the build process. You'll find the compiled binary in `bin/bootx64.efi`. Change that file to be named `bootia32.efi`.
> 1. Build my version of the GRUB boot loader, which has been modified to work with Enterprise. You can download the binary [at this link][2].
> 1. Extract the tarball and open a shell in the GRUB source directory.
> 1. Follow the steps detailed in the [`GRUB` file][3] at the top level of the Enterprise repository. Be sure to modify the paths so they work on your system.

If you have any additional questions please post. If you need coaching or for me to work you through the process, I can do that for a small fee. You can [contact me][4] for the details if this something that interests you.

[1]: https://github.com/SevenBits/Enterprise/releases
[2]: https://github.com/SevenBits/Enterprise/releases/download/v0.3.0/GRUB-modified.tar.gz
[3]: https://github.com/SevenBits/Enterprise/blob/master/GRUB
[4]: https://www.sevenbits.tk/contact.html

umma08, 2016-08-12:
Hi, i have this problem as well - but i am running a mid 2010 macbook pro, core2duo which runs el-capitan (10.11)

I am able to create the USB no problems, using the latest Kali distro, but on the drive select screen on boot the USB is not showing. 

Any ideas?

SevenBits, 2016-08-12:
@umma08 What are the contents of your USB drive?

umma08, 2016-08-14:
@SevenBits 

 ~/efi

~/efi/boot

~/efi/boot/boot.efi
~/efi/boot/boot.iso
~/efi/boot/bootx64.efi

SevenBits, 2016-08-14:
@umma08 Is your USB using a FAT32 filesystem?

umma08, 2016-08-14:
@SevenBits yes, the USB is a 16GB, and was formatted using Disk Utility. 

It was formatted MS-DOS(FAT) and as a Master Boot Record. 

I am trying now as a GUID Partition Map, to see if that makes a difference. 

I used both the latest Kali distro - downloaded through the MAC USB Loader app, and also downloaded manually through the Kali website (both were not seen on the boot screen) 

My choice of distribution was KALI.

umma08, 2016-08-14:
@SevenBits can confim: makes no difference if the disk is formatted as a GUID Partition Map.

cems2, 2016-08-14:
As perahps simple and obvious to you that is, That's a remarkable useful bit of valuable troubleshooting information to us lesser mortals :-)    One of the grand time syncs of trouble shooting is exhaustively enumerating all the possible binary alternatives.  (do I format it on a windows machine, linux or Mac;  try another USB stick;  32bit versus 64 bit, etc...).  You  may want to include that in the troubleshooting instructions.

SevenBits, 2016-08-14:
@umma08 Your Mac model should be capable of booting the boot loader. To be clear: when you hold down the Option key at startup, when the Mac’s boot selector shows up, you see your hard drive and the Recovery HD, but **not** your USB drive?

@cems2: I understand. This is not a very intuitive topic for most. :-) I’ll see what I can do about the documentation.

umma08, 2016-08-14:
@SevenBits Is might have something to do with this? SIP (System Integration Protection) that is on El-Capitan.

http://www.rodsbooks.com/refind/sip.html#disable

cems2, 2016-08-14:
It's not obvious to me why SIP would matter for a USB install.  SO I'm guessing that maybe you are suggesting that maybe SIP is the reason we can't get the thing to show the USB stick in the boot-options menu? 

That might make a lot of sense. 

But here's a couple possible things to consider if they consistent with this speculation.

   1. my computer is running Lion, not a newer OS.  Does Lion have SIP? 

    2.  If I install various distros on CDs  I still can't get them to boot properly.  It does try to use the CD, which is more than I can say for the USB stick, but it just fails to load from the CD.

Thus I like the idea that there is some master problem here that is preventing booting from anything but a blesses system.

My theory till now has been it's a matter of finding the right mix of 32-bit UEFI compatible loader, Linux distro, and USB stick voodoo.  But I've pretty much exahausted that.

The point I've reached is I'm afraid to try a hard drive install of linux, and afraid to try a Refied change to by boot loader, out of fear of bricking my computer.  Thus your application is what I want to get working.

A while back, when I was fooling around with an Orange Pi, I discovered that when I installed the distro onto a USB stick formatter by the mac it failed but it worked if I formatted the USB stick on windows or Rasberry Pi.  THis was despite choosing MBR and FAT 32 on the mac.  EMpirically these were not identical formatting operations.

http://www.rodsbooks.com/refind/sip.html#disable

umma08, 2016-08-14:
@cems2 @SevenBits 

just tried with SIP disabled. 

I rebooted into my System HD original clone, and then disabled the SIP using csrutil disable. 

you can check the status with csrutil status

i then rebooted into my cloned System HD and checked the status. SIP was disabled on the machine. 

i then booted using the option/alt - and the USB is still not detected. 

so this did not solved the problem.

SevenBits, 2016-08-14:
@umma08 @cems2 SIP is an operating-system level feature (mostly; the nuances are too complex to go into here). This has no bearing whatsoever on whether or not your USB can be booted by the system. I can, and do, boot USB drives with Linux on a system with SIP enabled.

umma08, 2016-08-14:
@cems2 @SevenBits Any other ideas of why my system won't recognise the USB? or any other options? I am going to try another version of Linux now - Ubuntu - to see if that makes a difference.

SevenBits, 2016-08-14:
@umma08 Without being able to re-produce the problem myself, there’s not much more I can do without more information.

umma08, 2016-08-14:
@SevenBits ok. is there anything i can do to give you more info?

cems2, 2016-08-14:
so ot sort of list out the binary set of choices here.

1. pre-2007 macs are UEFI  not EFI.  SHould this work for UEFI macs without doing something to change their firmware?  (and if there is a frimware change where to get it).

2.  there's an option in Mac-usb loader for using the distro's EFI or not.  (or something like that, the wording has changed in Mac-usb-loader over versions).  Should this be checked or unchecked under what cases?  I don't understand the explanation given.

3. Mac's with the UEFI loader are a 32 bit loader even though core 2 duo macs are also 64 bit CPU.  Thus one gets this moderate confusion if a 32 bit loader can load a 64 bit Linux.  The partial answer is that while it can of course do that at the OS boot itself, it may be that the EFI or UEFI loader that packaged with a 64 bit linux distro isn't compatible with the 32 bit UEFI firmware.  Or maybe it is.  Thus I think it's wise to try staying with a x386 version of Linux until one gets this to work, otherwise it's too confusing.

4.  Options for formatting the USB stick.  One is often in the position of running Mac-usb-loader on a newer machine (that it requires) than the target machine one plans to boot.  Is this a possible problem? Newer macs don't even have a GUI that allows anything but GUID formats

umma08, 2016-08-14:
@SevenBits @cems2 

I feel really stupid now, but after googling the issue, it seems i found the solution!

On the boot screen, all i had to do was take out the USB, and then reinsert it. 

The bootscreen then sees the USB as a bootable EFI drive. Booted up into Kali, no problems - but only after setting the nomodeset option in the 2) Advanced Settings. 

sorry about all that! hopefully this will help someone else in the same position. 

Have no idea why the bootscreen will not recognise the USB when it is plugged in on alt startup. 

It will only recognise it after reaching the bootscreen startup when taking out and putting back in the USB drive.

SevenBits, 2016-08-15:
@umma08 Glad you got it working.

umma08, 2016-08-16:
@SevenBits yes, me too! managed to get persistence up and running as well, as per the <a href="https://www.youtube.com/watch?v=mDRbTHCoj8U">youtube video </a> that you had linked to in another thread so all good to go now :)

cems2, 2016-08-16:
Sigh.   Did not work for me.

My macbook is a UEFI boot so maybe that’s the difference

SevenBits, 2016-08-16:
All MacBooks use UEFI booting...

evlPanda, 2020-06-10:
> On the boot screen, all i had to do was take out the USB, and then reinsert it.

You probably just saved me hours. Works for me : )

rotorrad, 2020-06-24:
Right, hey folks
I have an old mac book 2006 to be precise when I got it, I installed elementary os and the whole installation went smoothly. I had a time machine backup so wasn't worried about anything. I hated elementary OS so I decided to get rid of it. formated hdd on my other mac (much newer) and reinstalled the system from time machine backup. 
Cut the story short, I got no recovery partition now, and my MacBook doesn't see any USB drives in boot menu.
Any help would be great

evlPanda, 2020-06-24:
Might sound silly, but if you can’t see a USB drive in the boot menu, try just taking it out and putting it back in again. I had the same issue once.

Regards,

Michael Nitschke
m.nitschke@gmail.com
AF212E6D

rotorrad, 2020-06-24:
I have tried that, no luck. It's weird how one minute it was there and now is not. same laptop, same hdd, same ram.......
Thank you anyways