SevenBits/Mac-Linux-USB-Loader #125: USB Persistence doesn't work on SD

Patwd, 2016-08-15:
Seems USB Persistence doesn't see the SD Card ... 
This may not be a bug - since it's called USB persistence - not SD persistence - but I thought it would work for other removable media - as I need my USB ports for USB Ethernet card...

Selecting USB Persistence - only sees the USB drives / volumes ...

**Short Description of Problem:** put text here

**Expected Behavior:** 

put text here

**Actual Behavior:** 

put text here

**Steps to Reproduce:** 

put text here

*Note:* You can include screenshots and other media by dragging and dropping them onto this text area. Please do so, it helps solve your problem!

SevenBits, 2016-08-15:
@Patwd Does the installation wizard see your SD card?

aurelg, 2016-08-20:
Same here w/ Mint 18 on a regular USB stick, 2011 MBA.

The installation wizard sees the USB stick, and is able to create a casper-rw file at the root of it. However, filesystem changes are not persistent, no matter if I enable the ad hoc option in the boot menu by choosing 1/ "Modify Linux kernel boot options (advanced!)" 1/ Ubuntu and 5/ Persistency.

Interestingly the kernel command line (as shown by `dmesg|grep command`) shows no reference to `persistency`:

`BOOT_IMAGE=/casper/linux file=/preseed/ubuntu.seed boot=casper iso-scan/filename=/efi/boot/boot.iso quiet splash`

SevenBits, 2016-08-20:
And, to be clear, this works as expected on a USB stick?

aurelg, 2016-08-20:
@SevenBits no, it doesn't. Same symptoms than what @Patwd observed on a SD card.

SevenBits, 2016-08-20:
Now, that's odd. Did it work on previous versions of Ubuntu?

aurelg, 2016-08-20:
I didn't tried. Which ubuntu version did you tested it with? I may be able to give it a try within 24h.

aurelg, 2016-08-21:
OK, tried Ubuntu 14.04.5 and 15.10, still no luck. 

When starting 14.04.5, the kernel command line adds an option which might be connected to persistency, like `seed.persistent`, but it does not work. It's probably not relevant to try to fix that one since 16.04 is the latest LTS.

hkalan, 2016-08-28:
Hi

Persistence not working for me also:

Using Mac Linux USB Loader (3.3) I created Live USB (using a 8 GB Flash Drive), by using Linux image: "ubuntu-16.04.1-desktop-amd64.iso", after the installation was finished succesfully, I used "Persistence Manager" and created on the same USB drive 512 MB of persistence.

I used the USB to boot the system and used the boot menu (Enterprise) made sure that Persistence is enabled and booted successfully into Ubuntu, then I created a file on Desktop saved it and restarted the Ubuntu, I enabled Persistence again however after rebooting the file was not persisted, hence Persistence is not working as suppose to. Please assist me

Thanks

SevenBits, 2016-08-28:
@hkalan There is a bug in the current version of my boot loader. Please go into Setup USB Device > Edit Enterprise Configuration File, and paste the contents here.

hkalan, 2016-08-28:
@SevenBits 

Thanks for swift reply, following is  the content of enterprise.cfg: 

"#This file is machine generated. Do not modify it unless you know what you are doing.

entry Ubuntu
family Ubuntu"

Thanks

SevenBits, 2016-08-28:
@hkalan You're welcome, I happened to be on GitHub and I saw your message.

Try this (late here, so untested):

    entry Ubuntu
    family Ubuntu
    kernel /casper/vmlinuz.efi persistent

hkalan, 2016-08-28:
@SevenBits, I made the changes (below) and double checked to be sure the config is saved correctly, tested again as described in my initial comment, unfortunately persistence is not working in this case.
 
>entry Ubuntu
family Ubuntu
kernel /casper/vmlinuz.efi  #persisten
> 

Thanks for your support

SevenBits, 2016-08-28:
It looks like you did not spell 'persistent' correctly...

hkalan, 2016-08-28:
>
It looks like you did not spell 'persistent' correctly...

@SevenBits, It was copy/paste error here in the comment, it is pasted correctly in enterprise.cfg, thanks

SevenBits, 2016-08-28:
@hkalan Okay, I will look into this. Thank you.

hkalan, 2016-09-01:
> @hkalan Okay, I will look into this. Thank you.

@SevenBits  Could you please provide **an estimated release date or a workaround for this issue**, I really need persistence in order to be able to **use Ubuntu with my iMac Retina 27 (Late 2014)**. Persistence is required **to update the Ubuntu Graphic Card Driver for ATI card.** Currently a black (blank) screen is displayed and I can't do anything. However the Ubuntu has no problem loading in my MacBook Air (early 2013). Thanks you!

ajmichaels, 2016-09-01:
@SevenBits Persistence is also not working for me on a usb stick.  Followed all the instructions, set up persistence through the GUI, and enabled option 5 when booting.

SevenBits, 2016-09-01:
@ajmichaels This is a known bug. Please see #125.

SevenBits, 2016-09-01:
@hkalan These configuration file were confirmed by another person to work:

    entry Ubuntu
    family Ubuntu
    kernel /casper/vmlinuz.efi persistent

Can you confirm that you have setup persistence correctly?

ajmichaels, 2016-09-01:
@sevenbits I've set up persistence as you've outlined in the guides, and tried that config file, but it's still not working.

hkalan, 2016-09-01:
> @hkalan These configuration file were confirmed by another person to work:

>entry Ubuntu
family Ubuntu
kernel /casper/vmlinuz.efi persistent
Can you confirm that you have setup persistence correctly?

@SevenBits, Yes I can confirm Persistence file exists "casper-rw" and it is detected on the boot by enterprise, however after enabling Persistence in options menu (It is Highlighted Yellow) it does not work at all as mentioned in my original post.

Following is the screenshot of my enterprise.cfg file, therefore I can confirm it is setup correctly:

[image]

ajmichaels, 2016-09-01:
@sevenbits not sure if it helps, but when I use the config file you suggest, the loading screen says "boot parameters: persistentpersistent" instead of just "persistent".  When I remove the persistent flag from the config file but leave in the "kernel /casper/vmlinuz.efi", the boot parameters field becomes blank.

SevenBits, 2016-09-01:
@ajmichaels Yes, the spacing issue is something I'm aware of. I've fixed it in the source code, but it hasn't made its way into Mac Linux USB Loader yet.

Try this: keep that configuration (with the spacing issue) but select another option from before persistence, like `nomodeset`. See if, after selecting that, the persistence option stands alone. If so, then persistence should work.

ajmichaels, 2016-09-01:
@SevenBits That did it, thanks!

SevenBits, 2016-09-01:
@ajmichaels You’re welcome, happy to help.

hkalan, 2016-09-02:
> @ajmichaels Yes, the spacing issue is something I'm aware of. I've fixed it in the source code, but it hasn't made its way into Mac Linux USB Loader yet.

>Try this: keep that configuration (with the spacing issue) but select another option from before persistence, like nomodeset. See if, after selecting that, the persistence option stands alone. If so, then persistence should work.

@SevenBits, thanks to @ajmichaels I just noticed the spacing issue in Boot Parameters.

The current workaround for me is this (Enterprise 0.4.0 & Mac Linux USB Loader 3.3):

**In order to ENABLE persistence you should use Enterprise boot menu:**

- then select option `1) Boot Linux from ISO file`, the boot parameter `persistent` flag is automatically added by Enterprise when a persistence file is found, no need to do any extra steps.

**In order to DISABLE persistence, you  should use Enterprise boot menu:**

- select option `2) Modify Linux kernel boot options (advanced!)`

- Make sure option `5) Persistent - Make any changes to the flash storage persist` is highlighted yellow (i.e. selected). In this case boot parameters would become: `persistentpersistent` which is not the correct parameter for enabling Persistence; hence by enabling Persistence via boot options you would actually disable it.

NOTE: This is not expected behaviour and it is a bug in Enterprise

SevenBits, 2016-09-02:
@hkalan The boot parameter is not automatically added...

hkalan, 2016-09-02:
> hkalan The boot parameter is not automatically added...

@SevenBits Thanks, just noticed my mistake and updated the comment accordingly!

curiouscurious, 2016-11-24:
When I made my live USB, I left the box checked to skip the enterprise boot menu. And now I'm unable to access persistence. Is there any way to access initial boot menu on loading so that I can enable persistence? Or do I have to reformat the USB? And does one have to enable persistence on every reboot"?

curiouscurious, 2016-11-24:
This is my enterprise.cfg file if it helps. Perhaps I can modify the "auto boot 0"?... Also, why is persistent not mentioned anywhere in my kernel name? I thought it should be from the above discussion. I do in fact have a casper-rw on the USB.

autoboot 0
entry Kali
family Debian
kernel /live/vmlinuz findiso=/efi/boot/boot.iso boot=live noconfig=sudo username=root hostname=kali

SevenBits, 2016-11-26:
@curiouscurious Remove the whole `autoboot ...` line...

SevenBits, 2016-12-19:
Closing due to lack of activity.

closed 2016-12-19 by SevenBits