SevenBits/Mac-Linux-USB-Loader #127: No explanation for "Skip the boot selection menu" in the wiki

fivepapertigers, 2016-08-27:
Relatively trivial, but:

I am using your software for the first time to boot Ubuntu from a USB and there was an extra option "Skip the boot selection menu" that was not described in the wiki. Would love to see it explained so that I had a better sense of what my options are.

Installation still in progress, but I'm looking forward to the results!

SevenBits, 2016-08-28:
@fivepapertigers Thank you for the tip, I will take another look at the documentation.

For anyone wondering: this option tells the boot loader to skip showing the menu (I.e where you need to press 1 or 2) and just boot the distribution directly. This is useful for convenience, but the downside is that you cannot access the advanced boot settings menu to e.g enable persistence or disable kernel mode setting.

SevenBits, 2016-08-28:
@fivepapertigers [Updated](https://github.com/SevenBits/Mac-Linux-USB-Loader/wiki/General-Instructions).

closed 2016-08-28 by SevenBits