SevenBits/Mac-Linux-USB-Loader #129: "Grant Access" dialog should be more explicit about what needs to happen

kived, 2016-08-28:
**Short Description of Problem:**
Dialog says to just hit "Grant Access", but this doesn't actually explain what you need to do.

**Expected Behavior:** 
Hit "Grant Access", application continues with work

**Actual Behavior:** 
Dialog just appears again and again asking to "Grant Access"

**Steps to Reproduce:** 
Open application, choose to create Live USB, select an ISO and device, "Grant Access" dialog appears

After some Googling, I found a reference in #58 to hitting "Grant Access" while in the root of the USB device - navigating to the USB first worked. The dialog (or, more likely, a screen before the dialog is displayed) should note that you need to be in the root of the USB device. Perhaps the application is supposed to automatically open the dialog at the root of the USB device, and that is failing for some reason, but even if that's the case it would be best to inform the user in the event this happens.

kived, 2016-08-28:
In fact, on the second USB I created the dialog did automatically navigate to the USB. But the first one didn't, after multiple reinserts and restarts of the app.

SevenBits, 2016-08-28:
@kived Thank you, I will update the documentation.

closed 2016-08-28 by SevenBits